uaf/uaf #48: Implemented TranslateBrowsePathsToNodeIdsRequest() ?

guillaumelibert, 2015-01-22:
When I tried to use TranslateBrowsePathsToNodeIdsRequest() I get an error:
NotImplementedError: Wrong number or type of arguments for overloaded function 'ClientBase_processRequest'.
  Possible C/C++ prototypes are:
    uafc::Client::processRequest(uafc::ReadRequest const &,uafc::ReadResult &)
    uafc::Client::processRequest(uafc::AsyncReadRequest const &,uafc::AsyncReadResult &)
    uafc::Client::processRequest(uafc::WriteRequest const &,uafc::WriteResult &)
    uafc::Client::processRequest(uafc::AsyncWriteRequest const &,uafc::AsyncWriteResult &)
    uafc::Client::processRequest(uafc::MethodCallRequest const &,uafc::MethodCallResult &)
    uafc::Client::processRequest(uafc::AsyncMethodCallRequest const &,uafc::AsyncMethodCallResult &)
    uafc::Client::processRequest(uafc::BrowseRequest const &,uafc::BrowseResult &)
    uafc::Client::processRequest(uafc::BrowseNextRequest const &,uafc::BrowseNextResult &)
    uafc::Client::processRequest(uafc::CreateMonitoredDataRequest const &,uafc::CreateMonitoredDataResult &)
    uafc::Client::processRequest(uafc::CreateMonitoredEventsRequest const &,uafc::CreateMonitoredEventsResult &)
    uafc::Client::processRequest(uafc::HistoryReadRawModifiedRequest const &,uafc::HistoryReadRawModifiedResult &)

Do you have a working example for this request?

Regards,

WimPessemier, 2015-01-23:
Hi, ``processRequest(TranslateBrowsePathsToNodeIdsRequest(...))`` is indeed missing... you must be the first who wants to manually translate browsepaths (instead of letting the UAF do it automatically). :-)

I've implemented it for you, and there's also an [example](https://github.com/uaf/uaf/blob/master/examples/pyuaf/client/how_to_translate_browsepaths_manually.py) now. 
It's committed to both the master and long-term branch.

Let me know if you experience problems.
Cheers

guillaumelibert, 2015-01-23:
Wow that was fast,
I will keep you in touch with my testing.
Thanks!

guillaumelibert, 2015-03-16:
TranslateBrowsePathsToNodeIdsRequest is working.
Thanks

closed 2015-03-16 by guillaumelibert