uaf/uaf #50: Access to classic OPC DA server via ua wrapper from Linux

ulrichkroeger, 2015-02-12:
Hi Wim,
I have the following constellation under test:
- OPC DA Server (Matricon OPC Simulation)
- OPC UA Wrapper (Unified Automation)
both on the same Windows machine

- PyUAF client on Linux
I successfully build the PyUAF stuff on Linux and I started to run a simple example.
After instantiation of the client, I am receiving successfully a pyUAF client:
myClient = Client(ClientSettings("myClient", ["opc.tcp://139.22.44.126:21381"]))
returns:
<pyuaf.client.Client; proxy of <Swig Object of type 'uafc::Client *' at 0xa4b0290> >

Then I execute the read of the node:
    someAddress = Address(NodeId("iBucket Brigade.String",                 # NodeId identifier
                                 0),# NodeId namespace URI
                                 "urn:MatrikonOpcUaWrapper")                   # server URI
    # read the Value attribute of the node that we just addressed:
   result = myClient.read( [someAddress] )

After execution of the read, I am getting the following error:
ConnectionError: No connected session to invoke the service.

I believe that I don't know how to set correctly the node address to read from.
Here are my questions:
- Which namespace do I need to specify? Where do I get it from?
- For the server URI - do I need to specify the UA Wrapper or the underlying OPC DA server?

Additonal information:
I successfully accessed the OPC DA server from the UAExpert client on Windows machine - so the UA wrapper works fine.

Thanks very much for your help!
Regards,
Ulrich

WimPessemier, 2015-02-12:
Hello Ulrich,

Okay, your example code won't work as - indeed - you've specified the wrong namespace in your NodeId. 

I'll give you the full story (which I'll copy to the UAF docs some time):

A namespace is a string that provides a sort of "context" to your node. For instance:
  ``NodeId("tank.temperature", "http://my.company.com")`` 
is different from:
 ``NodeId("tank.temperature", "http://your.company.com")`` 

We both use the same NodeId identifier ("tank.temperature"), but the context is different (your tank.temperature is not the same as my tank.temperature).

Because the URI "http://your.company.com" is a lot of bytes that would need to be sent over the network every time you access the Node, OPC UA has thought of a clever thing: every server (yours too) has a Node called ``NamespaceArray``. Take a look at the NamespaceArray (under the ``Server`` Node) of your own server using UaExpert: this array maps integers (called indexes) to URIs. So instead of sending the whole namespace URI on every request to the server, OPC UA only sends these namespace indexes to the server. One thing to note is that 0 (the first item of the array) is ALWAYS reserved for "http://opcfoundation.org/UA/". So most of the standard nodes you see (like ``Server`` and ``NamespaceArray`` use this context.

So, what do you need to do? Use UaExpert and browse to your Node (iBucket Brigade.String). Using UaExpert, check the information of the NodeId: 
 - what is the Identifier (probably but not necessarily "iBucket Brigade.String")
 - what is the NamespaceIndex (e.g. 2)

Now you can create the NodeId like this: ``NodeId("iBucket Brigade.String", 2)``.

Keep in mind though that namespace indexes are not really "fixed", the server may one day map "http://your.company.com" to index 2, and the next day map it to index 3. So the most robust way is to identify your nodes using namespace URIs and not using namespace indexes. In the above case, we supposed the namespace index of your node is 2. So you can now read the NamespaceArray using UaExpert, and see which namespace URI (e.g. "http://matrikonblablabla.com" corresponds to index 2). Then create your NodeId like this:
  ``NodeId("iBucket Brigade.String", "http://matrikonblablabla.com")``.

pyUAF will automatically convert the URI "http://matrikonblablabla.com" to the index 2 before sending your request to the server. So your client will be robust, and you won't lose any performance.

About the serverUri: when you connect to the server using UaExpert, you'll see in the logging window something like
  ``ApplicationUri: 'urn:blablabla'``
This is your serverUri, which you need to include in the Address, so pyUAF knows which server hosts your node.
So finally, you can address a Node like this:

```python
Address(
      NodeId( "iBucket Brigade.String", "http://matrikonblablabla.com"),
      "urn:MatrikonOpcUaWrapper")
```

Let me know if it doesn't work,

Wim

ulrichkroeger, 2015-02-12:
Hi Wim,

thanks so much for your detailed explanations!
I first tried the "shortcut" method by specifying the namespace index, which is in fact "2" for "iBucket Brigade.String". It seems to be stable over time. 
Result: The same error message as before.

I also browsed for the NamespaceArray, as you explained. It has 8 elements!
See attached screenshot. 
Which Namespace do I need to specify?

I also had a closer look at the server uri: Here I am quite sure that it should have the suffix "MatrikonOpcUaWrapper", but in UaExpert I did not find the ApplicationUri in the log window, but a message called "found endpoint "..." - see screenshot. Do I need to specify the full endpoint or is the string 
[image]
"urn:MatrikonOpcUaWrapper" sufficient?

Thanks again for your help!!
Regards,
Ulrich
[image]

[image]

ulrichkroeger, 2015-02-12:
Hi Wim,
it is still not working - but one correction: I found the ApplicationUri, which is "urn://<myhostname>/MatrikonOpcUaWrapper". I tried to specify in my code as is but it did not work. I also tried the ip address instead of myhostname, without success - and of course I tried the urn "urn:MatrikonOpcUaWrapper" which did not work either.
Regards,
Ulrich

WimPessemier, 2015-02-12:
Hi Ulrich,

I've downloaded the demo version of the Matrikon wrapper and tested it myself. Apparently the wrapper exposes the server URI which you can see in the example code below. I've had to check the UAF logs to get this info as it wasn't obvious (in my interpretation of the OPC UA specs, this server URI should be the same as the application URI, which should be the same as the first item in the ``ServerArray`` node which is also exposed by the server). Anyway the code below works, I'll make sure that the future version of UAF adds more diagnostic capabilities to get a clearer view on connection issues like this.

Hope this works for you,
Cheers
Wim

```python
import pyuaf
from pyuaf.util             import Address, NodeId
from pyuaf.client           import Client
from pyuaf.client.settings  import ClientSettings

# create the client:
settings = ClientSettings("myClient", ["opc.tcp://localhost:21381"])
myClient = Client(settings)

# constants:
SERVER_URI = "urn:matrikon.com:MatrikonOpcUaWrapper"
NAMESPACE_URI = "opc.com://localhost/Matrikon.Opc.Simulation.1 (DA)"

# test: read 2 nodes and print the result:
node0 = Address(NodeId("iBucket Brigade.String", 2), SERVER_URI)
node1 = Address(NodeId("iBucket Brigade.UInt2", NAMESPACE_URI), SERVER_URI) # more robust
result = myClient.read( [node0, node1] )
print(result)
```

ulrichkroeger, 2015-02-12:
Hi Wim,
I am sure your solution solved one of my problems - so my server uri was not correct and is correct now. The behavior is now different: Before using your latest code, the read returned right away with the error message. Now, the read call seems to time out - and after a few seconds - unfortunately - I still receive the same error message - which of course does not tell me the real cause of the problem.
In difference to your test scenario, I use a Linux VM to connect to my Windows host, so my connection parameters are different to yours.
Here is my code:
[image]
Regards,
Ulrich

WimPessemier, 2015-02-12:
Hi Ulrich,

I've tested the very same thing now, and it's a bit tricky when using a VM. Your code basically times out as it cannot access the VM's OPC UA server. I guess you are using UaExpert inside the Windows VM?

This is how I got it to work:
 - using VirtualBox, I've changed the adapter type of my VM to "Bridged Adapter". The VM will now receive a "real" IP address from your router, just like a normal PC. In my case: 10.33.178.202. The hostname of my VM is ``wimpe-pc``.
 - I've added the following line in /etc/hosts:  ``10.33.178.202 wimpe-pc``. 
 - I've modified the code like this:
```python
import pyuaf
from pyuaf.util             import Address, NodeId
from pyuaf.client           import Client
from pyuaf.client.settings  import ClientSettings

# create the client:
settings = ClientSettings("myClient", ["opc.tcp://wimpe-pc:21381"])
# settings.logToStdOutLevel = pyuaf.util.loglevels.Debug # uncomment to enable logging 
myClient = Client(settings)

# constants:
SERVER_URI = "urn:matrikon.com:MatrikonOpcUaWrapper"
NAMESPACE_URI = "opc.com://localhost/Matrikon.Opc.Simulation.1 (DA)"

# test: read 2 nodes and print the result:
node0 = Address(NodeId("iBucket Brigade.String", 2), SERVER_URI)
node1 = Address(NodeId("iBucket Brigade.UInt2", NAMESPACE_URI), SERVER_URI) # more robust
result = myClient.read( [node0, node1] )
print(result)
```

If you can't get it to work, uncomment the line to enable the logging (see code) and please post the output somewhere (e.g. on http://pastebin.com).

If you can't configure adapter type of the VM, you may be able to set-up port/ip forwarding, I guess...

HTH,
Wim

ulrichkroeger, 2015-02-13:
Hi Wim,
good news: It works now! The only remaining problem I had was that the pyuaf Client Settings seem not to accept ip addresses, so I needed to use the host name instead and added this one into /etc/hosts.
Again - thank you very much for your detailed help and effort!
FYI: I am just doing a feasibility study and the current result leads into the direction that we might use pyuaf as our OPC client solution for connection to any type of OPS server, from Linux OS. If you are interested, I will keep you informed about this!
Regards,
Ulrich

WimPessemier, 2015-02-13:
Hi Ulrich, 
glad you got it to work!
And yes, I'd be grateful if you could keep me informed if you decide to use pyuaf for some project some day... It's interesting, and (career-wise) it's also nice for me to have a few "external" references of where the software I wrote is used :)
Thanks & regards,
Wim

closed 2015-02-13 by WimPessemier

ulrichkroeger, 2015-02-13:
Hi Wim,

of course I will do!
Thanks again for your help.

Regards,
Ulrich

abhinavrawat27, 2017-08-24:
Hi ulrichkroeger 

In your question, you have mentioned that you accessed the OPC DA server using UAexpert from windows using some UA wrapper. Can you explain it how you have done it? I have installed UAexpert but don't have any idea about uawrapper. I am using windows 10. 

Thanks
Abhinav

ulrichkroeger, 2017-08-24:
Hi Abhinav,
I used the OPC UA Wrapper from Matricon:
http://www.matrikonopc.com/opc-ua/products/ua-wrapper-com-opc-servers.aspx?_sm_byp=iVVsnvwVkkMvrWMQ
Regards,
Ulrich

abhinavrawat27, 2017-08-29:
Hi ulrichkroeger

As you said in your post, you tested pyuaf from linux and you had OPC DA server running on windows machine so used OPC UA wrapper on windows. In my case I have a OPC DA server which is running on a machine and I need to access it from linux. Is there any UA wrapper which I can install in linux and then try pyuaf to access the OPC DA server.

Thanks
Abhinav

ulrichkroeger, 2017-08-29:
Hi Ahinav,
I am sorry, I do not know - I recommend you to search via Google if you can find it.
BTW you mentioned "In my case I have a OPC DA server which is running on a machine" - which machine? Linux? Usually OPC DA Server is running on Windows - thus, the UA wrapper can also run on Windows => then it is exactly the sceanrio I tested, where I used the  Matricon UA wrapper.

abhinavrawat27, 2017-08-29:
Hi ulrichkroeger

There is Moulding machine on which this OPC DA server is running. That's why I do not if its windows or any other platform. Also I cannot configure it, so cannot install ua wrapper on it. That's why I am looking for a ua wrapper for linux but I don't think its available in case of linux. 
Do you know any other way of accessing DA server from linux.

Thanks
Abhinav

ulrichkroeger, 2017-08-29:
Hi Abhinav,
no, sorry, that was my only actvity in this area.
Regards,
Ulrich