uaf/uaf #51: Build breaks with OPC UA SDK 1.4.3

robi-wan, 2015-02-26:
Hi Wim,

I tried to build uaf (on Windows with VS 2010) with the current version of OPC UA SDK 1.4.3 - with errors:
```
[CL] helperfunctions.cpp
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(91, 0): warning C4005: 'AF_IPX' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(460) : see previous definition of 'AF_IPX'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(131, 0): warning C4005: 'AF_MAX' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(479) : see previous definition of 'AF_MAX'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(168, 0): warning C4005: 'SO_DONTLINGER' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(402) : see previous definition of 'SO_DONTLINGER'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(212, 0): error C2011: 'sockaddr' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(485) : see declaration of 'sockaddr'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(390, 0): error C2059: syntax error : 'constant'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(390, 0): error C3805: 'constant': unexpected token, expected either '}' or a ','
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(524, 0): warning C4005: 'IN_CLASSA' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(287) : see previous definition of 'IN_CLASSA'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(530, 0): warning C4005: 'IN_CLASSB' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(293) : see previous definition of 'IN_CLASSB'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(536, 0): warning C4005: 'IN_CLASSC' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(299) : see previous definition of 'IN_CLASSC'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(547, 0): warning C4005: 'INADDR_ANY' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(304) : see previous definition of 'INADDR_ANY'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(549, 0): warning C4005: 'INADDR_BROADCAST' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(306) : see previous definition of 'INADDR_BROADCAST'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\ws2def.h(583, 0): error C2011: 'sockaddr_in' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(312) : see declaration of 'sockaddr_in'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(132, 0): error C2011: 'fd_set' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(68) : see declaration of 'fd_set'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(167, 0): warning C4005: 'FD_SET' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(102) : see previous definition of 'FD_SET'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(176, 0): error C2011: 'timeval' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(111) : see declaration of 'timeval'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(232, 0): error C2011: 'hostent' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(167) : see declaration of 'hostent'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(245, 0): error C2011: 'netent' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(180) : see declaration of 'netent'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(252, 0): error C2011: 'servent' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(187) : see declaration of 'servent'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(264, 0): error C2011: 'protoent' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(199) : see declaration of 'protoent'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(360, 0): error C2011: 'WSAData' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(322) : see declaration of 'WSAData'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(458, 0): error C2011: 'sockproto' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(494) : see declaration of 'sockproto'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(500, 0): error C2011: 'linger' : 'struct' type redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(531) : see declaration of 'linger'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(513, 0): warning C4005: 'SOMAXCONN' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(544) : see previous definition of 'SOMAXCONN'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(542, 0): warning C4005: 'FD_READ' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(562) : see previous definition of 'FD_READ'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(545, 0): warning C4005: 'FD_WRITE' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(563) : see previous definition of 'FD_WRITE'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(548, 0): warning C4005: 'FD_OOB' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(564) : see previous definition of 'FD_OOB'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(551, 0): warning C4005: 'FD_ACCEPT' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(565) : see previous definition of 'FD_ACCEPT'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(554, 0): warning C4005: 'FD_CONNECT' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(566) : see previous definition of 'FD_CONNECT'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(557, 0): warning C4005: 'FD_CLOSE' : macro redefinition
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(567) : see previous definition of 'FD_CLOSE'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(1578, 0): error C2375: 'accept' : redefinition; different linkage
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(742) : see declaration of 'accept'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(1600, 0): error C2375: 'bind' : redefinition; different linkage
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(747) : see declaration of 'bind'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(1619, 0): error C2375: 'closesocket' : redefinition; different linkage
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(752) : see declaration of 'closesocket'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(1638, 0): error C2375: 'connect' : redefinition; different linkage
[CL]         C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock.h(754) : see declaration of 'connect'
[CL] C:\Programme\Microsoft SDKs\Windows\v7.0A\include\winsock2.h(1659, 0): error C2375: 'ioctlsocket' : redefinition; different linkage
[... 178 lines cut ...]
Another simple solution also worked: removing the line which contains '#include <windows.h>'  - as this is the only line for the if-branch of the ifdef the whole ifdef can be changed:

```
 src/uaf/util/helperfunctions.h | 4 +---
 1 file changed, 1 insertion(+), 3 deletions(-)

diff --git a/src/uaf/util/helperfunctions.h b/src/uaf/util/helperfunctions.h
index 455156d..1bd58de 100644
--- a/src/uaf/util/helperfunctions.h
+++ b/src/uaf/util/helperfunctions.h
@@ -25,9 +25,7 @@
 #include <string>
 #include <algorithm>
 
-#ifdef _WIN32
-#include <windows.h>
-#else
+#ifndef _WIN32
 #include <limits.h>
 #include <unistd.h>
 #endif
```

At least it compiles.
Just tested under windows; I did not run the unit tests.

WimPessemier, 2015-02-26:
Hi,

Ah sorry I did not find the time to test the new SDK, thanks a lot for investigating and sharing solutions!
I'll look at it over the weekend, but it seems like you already did all the hard work :-)

Thanks,
Wim

robi-wan, 2015-02-26:
I forgot to mention: I compiled uaf with these fixes just against SDK 1.4.3 - I don't know if it also works (compiles) with older SDK 1.4.* versions.

WimPessemier, 2015-03-02:
Hi,

I've applied your last solution (i.e. remove ``#include <windows.h>``) and tested it on Linux and Windows with SDK v1.4.2 and v1.4.3. Everything appears to be working so I've updated the master and long-term branch to include the bugfix.

Thanks again!

Wim

closed 2015-03-02 by WimPessemier