uaf/uaf #54: Question regarding destruction of Clients

Atanamo, 2015-03-03:
Hi,
a question regarding the life cycle of a UA client:

Is there a way to explicitly destroy a client?
Or do I need to do this anyway, if I want to discard a client?

The use case I have is as follows:
* Own class X derives from the Pyuaf Client class to have a custom client
* Creating of a new instance of X (It is not monitoring nor calling/reading any node)
* Check, if instance/client can browse a special node (Done by an internal routine, only the custom client knows of)
** If the check succeeds, the custom client is added to a global list to provide its service to my app
** If not, the client should be discarded. It is not relevant for further processing...

By now, I'm simply not adding the client to the global list, if it should be discarded. But I wonder, if the client's object does stay in memory or anything else. I'm afraid, it cannot be garbage collected automatically, without calling a kind of destructor?

Hope, you can help me, thanx :-)

WimPessemier, 2015-03-04:
Hi,
I think indeed it's better to delete the client manually if its check fails:
```python
del myClient
```
I'm doing this as well for the unit tests: every test should have a "fresh" client so at the end of each test the client gets deleted (see [example](https://github.com/uaf/uaf/blob/master/unittests/pyuaftests/client/client_browse.py)).

If you want to verify if a client gets garbage collected, an easy way may be to enable the logging. You'll see a *lot* of output, but when a client is deleted (manually or by garbage collection), it will be visible in the logging:
```
>>> import pyuaf
>>> from pyuaf.client import Client
>>> from pyuaf.client.settings import ClientSettings
>>> settings = ClientSettings()
>>> settings.discoveryUrls.append("opc.tcp://localhost:4841")
>>> settings.logToStdOutLevel = pyuaf.util.loglevels.Debug
>>> myClient = Client(settings)
2015-03-04 19:07:29.047 Info                      LoggerFactory       Changing stdout log level to Debug
2015-03-04 19:07:29.047 Debug                     Client              Thread has started to periodically check the requests
2015-03-04 19:07:29.047 Debug                     Client              The discoveryUrls were changed, so we rediscover the system
2015-03-04 19:07:29.047 Debug                     Discoverer          Finding all configured servers
2015-03-04 19:07:29.048 Debug                     Discoverer          Finding the servers for URL 'opc.tcp://localhost:4841' (timeout 2000ms)
2015-03-04 19:07:29.049 Error                     Discoverer          Could not find servers at 'opc.tcp://localhost:4841' [OPC-UA:BadDisconnect]
2015-03-04 19:07:29.049 Error                     Discoverer          The FindServers service failed for these URL(s): 'opc.tcp://localhost:4841' (BadDisconnect)
>>> del myClient
2015-03-04 19:07:32.811 Debug                     Client              Destructing the client
2015-03-04 19:07:32.850 Debug                     Resolver            Destructing the resolver
2015-03-04 19:07:32.850 Debug                     Resolver            Now destructing the logger
2015-03-04 19:07:32.850 Debug                     SessionFactory      Destructing the SessionFactory
2015-03-04 19:07:32.850 Debug                     SessionFactory      Deleting all sessions and their subscriptions and monitored items
2015-03-04 19:07:32.850 Debug                     SessionFactory      All sessions have been deleted
2015-03-04 19:07:32.850 Debug                     SessionFactory      Now destructing the logger
2015-03-04 19:07:32.851 Debug                     Discoverer          Destructing the discoverer
2015-03-04 19:07:32.851 Debug                     Discoverer          Now destructing the logger
2015-03-04 19:07:32.851 Debug                     AddressCache        Destructing the address cache
2015-03-04 19:07:32.851 Info                      AddressCache        Clearing the address cache
2015-03-04 19:07:32.851 Debug                     AddressCache        Now destructing the logger
2015-03-04 19:07:32.851 Debug                     MonEvtsReqStore     Now clearing the whole store
2015-03-04 19:07:32.851 Debug                     MonEvtsReqStore     Now destructing the logger
2015-03-04 19:07:32.851 Debug                     MonDataReqStore     Now clearing the whole store
2015-03-04 19:07:32.851 Debug                     MonDataReqStore     Now destructing the logger
2015-03-04 19:07:32.851 Debug                     Client              Now destructing the logger
2015-03-04 19:07:32.851 Info                      LoggerFactory       The logger factory was destructed
```

Anyway if it would be possible to redesign your application so that you don't have to create-test-destruct clients all the time, it would be better I guess (perhaps it can be done with a pool of clients that accept a callback function which implements your check?). But manually deleting the clients shouldn't lead to memory leaks in principle.

Hope this helps!

Atanamo, 2015-03-05:
Ah ok, that's all I wanted to know :-)
Thanx very much!

closed 2015-08-29 by WimPessemier