uaf/uaf #55: Make error on Raspbian

MKetonen, 2015-03-29:
Hi,

I'm trying to compile uaf on Raspberry Pi with Raspbian image from Codesys (http://store.codesys.com/codesys-control-for-raspberry-pi-sl.html) but make command gets an error on libuafutil.so.

Here is the cmake:

pi@raspberrypi ~/Downloads/uaf/build $ cmake ../src -DUASDK=/opt/unifiedautomation -DCMAKE_CXX_FLAGS="-fPIC"
-DCMAKE_C_FLAGS="-fPIC" -DPYTHON_INCLUDE_DIR=/usr/lib/Python3.2
-DBUILD_SHARED_LIBS=ON
-- The specified Unified Automation SDK directory was found
-- OK, the SDK has version 1.4 or higher
-- The UaServerCPP demo server was found, so the unit tests can be run.
-- Including Python include path: /usr/lib/Python3.2
-- The following directories are included:
--  * /opt/unifiedautomation/include
--  * /opt/unifiedautomation/include/uabase
--  * /opt/unifiedautomation/include/uapki
--  * /opt/unifiedautomation/include/uastack
--  * /usr/include
--  * /home/pi/Downloads/uaf/src
-- * /usr/lib/Python3.2
--
-- ========== SUMMARY ==========
-- SDK was found    : TRUE (/opt/unifiedautomation)
-- SWIG was found   : TRUE (/usr/bin/swig2.0)
-- Python was found : TRUE (/usr/lib/python2.7/config/libpython2.7.so)
-- Options          : UASTACK_WITH_HTTPS = ON
-- This means that the UAF will be built, including Python wrappers
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:
CMAKE_C_FLAGS

-- Build files have been written to: /home/pi/Downloads/uaf/build

And the make goes 27% and then an error like this:

Linking CXX shared library /home/pi/Downloads/uaf/lib/libuafutil.so
/usr/bin/ld: skipping incompatible
/opt/unifiedautomation/lib/libuabase.a when searching for -luabase
/usr/bin/ld: cannot find -luabase
/usr/bin/ld: skipping incompatible
/opt/unifiedautomation/lib/libuapki.a when searching for -luapki
/usr/bin/ld: cannot find -luapki
/usr/bin/ld: skipping incompatible
/opt/unifiedautomation/lib/libuastack.so when searching for -luastack
/usr/bin/ld: cannot find -luastack
/usr/bin/ld: skipping incompatible
/opt/unifiedautomation/lib/libxmlparser.a when searching for -lxmlparser
/usr/bin/ld: cannot find -lxmlparser
collect2: ld returned 1 exit status
uaf/util/CMakeFiles/uafutil.dir/build.make:1163: recipe for target '/home/pi/Downloads/uaf/lib/libuafutil.so' failed
make[3]: *** [/home/pi/Downloads/uaf/lib/libuafutil.so] Error 1
CMakeFiles/Makefile2:93: recipe for target 'uaf/util/CMakeFiles/uafutil.dir/all' failed
make[2]: *** [uaf/util/CMakeFiles/uafutil.dir/all] Error 2
CMakeFiles/Makefile2:155: recipe for target 'uaf/client/CMakeFiles/uafclient.dir/rule' failed
make[1]: *** [uaf/client/CMakeFiles/uafclient.dir/rule] Error 2
Makefile:119: recipe for target 'uafclient' failed
make: *** [uafclient] Error 2

I'm a total noob on Linux so detailed advice would be great!

Thank you!

Mauri Ketonen
Alastaro
Finland

WimPessemier, 2015-03-30:
Hi Mauri,

Is your Unified Automation SDK precompiled for the architecture of your Raspberry Pi? 
``/usr/bin/ld: skipping incompatible`` basically says that the correct library (e.g. libuapki.a) has been found **but** the UAF cannot be linked to it because this library was not compiled for your specific hardware (e.g. x86, x86-64, ARM, ...). 

Don't know if it is applicable to your situation, but AFAIK, Unified Automation only provides a demo SDK for x86 (while you'd need one for ARM, I guess).

Maybe you can contact Unified Automation to see if they have a compatible SDK?

If I'm making false assumptions, please correct me...

Cheers,

Wim

MKetonen, 2015-04-01:
Hi!

Seems U are right! I'm waiting for a CD to arrive so I can test the ARM compatible SDK!

Thank You!

-Maukka

closed 2015-08-29 by WimPessemier