uaf/uaf #56: Access to Changes in GeneralModelChangeEvent

jorrita, 2015-04-14:
This may be a RTFM question, but I can't seem to figure it out:
How do I get access to the Changes that belong to a GeneralModelChangeEvent?

According to the documentation, Changes should be a property of GeneralModelChangeEventType and contain an array with the affected nodes.

I tried the following (using the server_cpp_demo sample):

address_Server = Address( NodeId(pyuaf.util.opcuaidentifiers.OpcUaId_Server, 0), "urn:UnifiedAutomation:UaServerCpp" )
eventFilter = EventFilter()
eventFilter.selectClauses.resize(1)
eventFilter.selectClauses[0].attributeId = pyuaf.util.attributeids.Value
eventFilter.selectClauses[0].browsePath.append(QualifiedName("Changes", 0))
eventFilter.selectClauses[0].typeId = NodeId(pyuaf.util.opcuaidentifiers.OpcUaId_GeneralModelChangeEventType, 0)
res = meynClient.createMonitoredEvents(addresses=[address_Server], eventFilter=eventFilter, notificationCallbacks=[cb])

What i get back in my callback looks wrong:
>>> mynotification
<pyuaf.client.EventNotification( - clientHandle       : 0
                                 - fields[]
                                    - fields[0]       : This array type cannot be displayed)>
>>> mynotification.fields[0]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/maarten/work/python/uaf/lib/pyuaf/util/__init__.py", line 1753, in __getitem__
    def __getitem__(self, *args): return ___init__.VariantVector___getitem__(self, *args)
TypeError: Invalid type!

WimPessemier, 2015-04-14:
Hi, I don't have experience with the model change events, but it seems that the Changes attribute is a special type (UaModelChangeStructureDataType) which is not wrapped by the UAF. So the ``"Invalid type!"`` is not a good error description, it should actually say that the type is not supported the UAF. Implementation won't be so difficult at first sight:
 - first of all the error message should be made more clear
 - then UaExtensionObject should to be wrapped
 - finally UaModelChangeStructureDataType should be wrapped.

I don't think it's a lot of work, but I'm busy with other parts of the framwork right now. I'll try to implement your request in the coming days.

WimPessemier, 2015-04-21:
Hi, I've implemented the things mentioned above. So pyuaf.util also contains a class ExtensionObject and ModelChangeStructureDataType now.
I've also added [an example](https://github.com/uaf/uaf/blob/master/examples/pyuaf/client/how_to_monitor_model_changes.py) so you can easily find out if this suits your needs.

Let me know if this doesn't solve your problem,
Wim

jorrita, 2015-04-22:
Perfect, that is exactly what I was looking for!
Jorrit

closed 2015-04-22 by jorrita