uaf/uaf #61: Virtualenv support

robi-wan, 2015-05-27:
Hi Wim,

thanks (again) for this great library.

We use uaf in several internal projects. Currently I was forced to try a newer version of uaf (in our main app we still use uaf 1.5).
Your instructions on installing uaf to use it from python use a *global* method whereas I needed a more localized variant. A common way to separate the python libraries for different projects is to use [virtualenv](https://virtualenv.pypa.io/en/latest/). But using the environment variable *PYTHONPATH* counteracts with this approach.

But I found a way to use uaf in a virtualenv.

1. Build a current version of uaf. I use Windows (XP/Server 2003), cmake 3.2.2, Python 2.7.5, SWIG 3.0.5. I removed the .cxx files from the lib
   directory.
2. Copy the contents of the `lib` folder into a virtualenv under `<virtualenv>/Lib/site-packages/pyuaf`. The folder structure should look like this:

    ```
    Lib/site-packages/pyuaf
    +-- pyuaf
        +-- client
        +-- util
    ```

3. Delete the file `<virtualenv>/Lib/site-packages/pyuaf/__init__.py`.

  When Python starts it scans all folders under `<virtualenv>/Lib/site-packages/` and searches for python packages via an `__init__.py` file.
  When it finds `<virtualenv>/Lib/site-packages/pyuaf/__init__.py` it thinks this is the root of a new package.
  But this would require all (py)uaf packages to be imported like `import pyuaf.pyuaf.client` this does not match the internal package structure.
  Therefore: I deleted this file.
  (It is not necessary even when using the `PYTHONPATH` environment variable approach to use uaf.)

4. Copy the OPC UA libraries to `<virtualenv>/Lib/site-packages/pyuaf`. For the OPC UA SDK v1.4 under Windows these are:
  * libeay32.dll
  * libxml2.dll
  * ssleay32.dll
  * uastack.dll

  The first three libraries are open source and free to distribute (I hope), the uastack.dll [comes](http://documentation.unified-automation.com/uasdkc/1.4.0/readme.txt) with a ["The Reciprocal Community License"](https://opcfoundation.org/license/rcl/1.00/opc-reciprocal-community-license-1.00.pdf).

5. Now the fun part: how does python knows there is a new package in town: [.pth] (https://docs.python.org/2/library/site.html) [files](http://bob.ippoli.to/archives/2005/02/06/using-pth-files-for-python-development/).
   http://lucumr.pocoo.org/2012/6/22/hate-hate-hate-everywhere/ (section "PTH: The failed Design that enabled it all")
   I took the inspiration for this hack from http://sourceforge.net/p/pywin32/bugs/669/ and https://gist.github.com/pfmoore/fd09fff11b4fa5f8cb25#file-pywin32_wheel-py-L40.
   I created a .pth file `<virtualenv>/Lib/site-packages/pyuaf.pth` with the following content:

    ```
# .pth file for pyuaf
pyuaf
# Add pyuaf folder to front of PATH
import os;os.environ["PATH"]=(os.path.join(sitedir,"pyuaf")+';'+os.environ["PATH"]);
    ```

  This adds the folder `<virtualenv>/Lib/site-packages/pyuaf` to the PATH environment variable and ensures that the libraries (from the step above) are found.
  (I just tested this under Windows - but I hope for linux support it is sufficient to change the .pth file to use `LD_LIBRARY_PATH` instead of `PATH`.)

After these steps I can use multiple versions of uaf on my computer (each version in another virtualenv).

I even pushed this a little bit further and created a portable python (inspired by https://github.com/pericazivkovic/portablepython) with all required python packages for an internal application pre-installed so that the deployment is easier. To check the created portable python I used a script similar to https://github.com/pericazivkovic/portablepython/blob/master/2.7/test_imports.py.
Would it be possible to add a way to check the (py)uaf version from python? E.g. adding a `__version__` attribute somewhere or copy the uaf/version.txt to the lib folder during compilation? 
(Portable Python [uses this approach](https://github.com/pericazivkovic/portablepython/blob/master/2.7/test_imports.py#L28-29) to check the version of pywin32.)

Maybe someday there can also be a (py)uaf package as a [wheel](http://wheel.readthedocs.org/en/latest/) an PyPI...

I would like to ask you to do the following changes:
* Remove the file `uaf/src/__init__.py` (that is the one I deleted in step 3.).
* Copy the needed libraries from the OPC UA SDK during compilation to the lib folder (see https://github.com/uaf/uaf/issues/23#issuecomment-36731717)
* Add a way to check the uaf version from python.

WimPessemier, 2015-05-29:
Hi,

Thanks a lot for the detailed explanation, instructions and links! I think anything that may simplify distribution and installation is very very welcome in the case of (Py)UAF :-)

I can easily do the changes you ask, as they are very clear. But I'm unfamiliar with "pth files", wheel, and I rarely use virtualenv, ... so it's hard for me to give comments on that part. If you feel anything is useful to add to the framework (like pth files), please do. Also if you'd like to have better access to the main uaf repository (i.e. github write/admin access) just let me know.

The current master branch is something like a "transition" version until I can (finally) release v2.0.0. There's not much more I'd like to add to that version, I just need to find some time to finish it :-). I will certainly include your feature requests before releasing v2.0.0. I hope from that version onwards, the API can remain much more stable than it is now... It should also be easier to apply feature requests and bug fixes much faster, as I don't have to spend time on re-writing parts of the framework anymore :-).

Cheers,

Wim

WimPessemier, 2015-06-03:
The 3 changes you mentioned are done, see https://github.com/uaf/uaf/pull/68/files.

Let me know in case something doesn't work as expected,

Wim

closed 2015-06-03 by WimPessemier

robi-wan, 2015-06-03:
Hi Wim,

thanks for the fixes. I did not test them all (actually I am currently not at work...) but one our Continuos Integration server already caught:

````
[16:48:00][Step 1/4] -- Trying to copy the SDK libraries (set COPY_SDK_LIBS to OFF if you dont want this)
[16:48:00][Step 1/4] CMake Error at UafMacros.cmake:430 (if):
[16:48:00][Step 1/4]   Syntax error in cmake code at
[16:48:00][Step 1/4] 
[16:48:00][Step 1/4]     C:/BuildAgent/work/4e5aa514b5bf26c9/src/UafMacros.cmake:430
[16:48:00][Step 1/4] 
[16:48:00][Step 1/4]   when parsing string
[16:48:00][Step 1/4] 
[16:48:00][Step 1/4]     c:\development\tools\opc_ua_sdk_14/bin/uastack.dll
[16:48:00][Step 1/4] 
[16:48:00][Step 1/4]   Invalid escape sequence \d
[16:48:00][Step 1/4] Call Stack (most recent call first):
[16:48:00][Step 1/4]   UafMacros.cmake:451 (copyFile)
[16:48:00][Step 1/4]   CMakeLists.txt:73 (copySdkLibraries)
[16:48:00][Step 1/4] 
[16:48:00][Step 1/4] 
[16:48:00][Step 1/4] -- Configuring incomplete, errors occurred!
````

Regarding this [post](http://stackoverflow.com/questions/14112279/cmake-parse-error-invalid-escape-sequence-o) changing the macro into a function should work.
This would mean changing [copyFile](https://github.com/uaf/uaf/blob/master/src/UafMacros.cmake#L428-L437) into a function should fix this:

````
 src/UafMacros.cmake | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/src/UafMacros.cmake b/src/UafMacros.cmake
index 0892388..0c8e1b7 100644
--- a/src/UafMacros.cmake
+++ b/src/UafMacros.cmake
@@ -423,9 +423,9 @@ ENDMACRO(setPyUafTargetProperties)
 
 # ----------------------------------------------------------------------------
 # copyFile(file, to)
-#    This macro will copy the given file to the given destination.
+#    This function will copy the given file to the given destination.
 # ----------------------------------------------------------------------------
-MACRO(copyFile _FILE _TO)
+FUNCTION(copyFile _FILE _TO)
 
     if( EXISTS "${_FILE}" )
         MESSAGE(STATUS "Copying ${_FILE} to ${_TO}")
@@ -434,7 +434,7 @@ MACRO(copyFile _FILE _TO)
         MESSAGE(FATAL_ERROR "Copying ${_FILE} failed: file not found!")
     endif( EXISTS "${_FILE}" )
 
-ENDMACRO(copyFile)
+ENDFUNCTION(copyFile)
 
 # ----------------------------------------------------------------------------
 # copySdkLibraries()
````

Works under Windows Server 2003 with VS 2010.

WimPessemier, 2015-06-03:
Hi,

Thanks for reporting the bug and figuring out the solution!
On my WinXP test platform it worked as expected though, maybe because I'm using the TortoiseGit bash shell (accepting both \ and / separators) instead of cmd prompt.

Anyway perhaps more of the CMake macro's should rather be CMake functions, I'll have a look at that.

Thanks,

Wim