uaf/uaf #62: Crash when calling method on server developed with Prosys OPC UA Java SDK

daveneijsen, 2015-06-01:
Hi Wim,

First of all I would like to say that we are using PyUAF in combination with the OPC UA C++ SDK and it works without problems.

Recently I started to use PyUAF as a test client for an OPC UA server developed with the Prosys OPC UA Java SDK. I've used the example server (SampleConsoleServer) from the SDK. This server creates a basic address space with amongst others a 'MyDevice' node with a method 'MyMethod'. The method has two input arguments: a String and a Double. The output argument is a Double and contains the result of the specified math operation (sin, cos, etc) applied to the given input argument of type Double.

The 'MyMethod' function can be called successfully from UA Expert. When calling it with PyUAF I'm running into an error. Described below are some test scenarios for the 'MyMethod' function.

**Test scenarios**
1) Synchronous call with valid input arguments; success
```python
    result = myClient.call(parent, method, inputArgs=["sin",Double(0)])
```
2) Synchronous call with invalid input arguments; **failure**
```python
    result = myClient.call(parent, method, inputArgs=[])
```
This call seems to crash PyUAF. Below is the last couple of lines of debug output:
```
 2015-05-29 17:16:23.795 Debug   myClient        SessionFactory      Copying the session information to the invocation
 2015-05-29 17:16:23.795 Debug   myClient        Session-0           Fetching session information:
 2015-05-29 17:16:23.795 Debug   myClient        Session-0            - clientConnectionId          : 0
 2015-05-29 17:16:23.795 Debug   myClient        Session-0            - sessionState                : 1 (Connected)
 2015-05-29 17:16:23.795 Debug   myClient        Session-0            - serverUri                   : urn:prosysopc.com:OPCUA:SampleConsoleServer
 2015-05-29 17:16:23.795 Debug   myClient        Session-0            - lastConnectionAttemptStep   : ActivateSession
 2015-05-29 17:16:23.795 Debug   myClient        Session-0            - lastConnectionAttemptStatus : Good
 2015-05-29 17:16:23.795 Debug   myClient        SessionFactory      Forwarding the invocation to session 0
 2015-05-29 17:16:23.795 Debug   myClient        Session-0           Invoking the service at the session level
 2015-05-29 17:16:23.795 Debug   myClient        Session-0           Copying the data from the synchronous UAF request to the SDK level
 2015-05-29 17:16:23.795 Debug   myClient        Session-0           Invoking the synchronous request at the SDK level
 2015-05-29 17:16:23.796 Debug   myClient        Session-0           Copying the data from SDK level to the UAF result
 terminate called after throwing an instance of 'std::length_error'
   what():  vector::_M_fill_insert
 Aborted (core dumped)
```
3) Asynchronous call with valid input arguments; success
```python
    def printMyAsyncCallResult(asyncResult):
        print(" --> received async method call result:")
        print(asyncResult)
    result = myClient.beginCall(objectAddress = parent,
                                methodAddress = method,
                                inputArgs     = ["sin",Double(0)],
                                callback      = printMyAsyncCallResult )
```
```
 ==================================================================
  - overallStatus                : Good
  - requestHandle                : 2
  - targets[]
     - targets[0]
        - clientConnectionId     : 0
        - status                 : Good
        - opcUaStatusCode        : 0
        - outputArguments[]      : [(Double)0]
 ==================================================================
```
4) Asynchronous call with invalid input arguments; success
```python
    result = myClient.beginCall(objectAddress = parent,
                                methodAddress = method,
                                inputArgs     = [],
                                callback      = printMyAsyncCallResult )
```
```
 ==================================================================
 - overallStatus                : The async call was completed with error: BadArgumentsMissing
 - requestHandle                : 1
 - targets[]
    - targets[0]
       - clientConnectionId     : 0
       - status                 : The server could not successfully process the MethodCall service: BadArgumentsMissing
       - opcUaStatusCode        : 2.15522e+09
       - outputArguments[]      : []
 ==================================================================
```
**Additional test scenarios**
I've experienced the same crash with another OPC UA server I've developed, not the default server example from the SDK. This server has a method with 1 input argument of type ByteString, and 2 output arguments: NodeId and UInt32. Below are the results of calling this method:

1) Synchronous call with valid input parameters; success
```python
    result = myClient.call(parent, method, inputArgs=[ByteString()])
```
2) Synchronous call with invalid input parameters; **failure**
```python
    result = myClient.call(parent, method, inputArgs=[])
```

Another method has 2 input arguments of type UInt32, and no output arguments. Below are the results of calling this method, both synchronous and asynchronous.

1) Synchronous call with valid input arguments; **failure**
```python
    result = myClient.call(parent, method, inputArgs=[UInt32(1),UInt32(1)])
```
2) Synchronous call with invalid input arguments; **failure**
```python
    result = myClient.call(parent, method, inputArgs=[])
```
3) Asynchronous call with valid input arguments; success
```python
    result = myClient.beginCall(objectAddress = parent,
                                methodAddress = method,
                                inputArgs     = [UInt32(1),UInt32(1)],
                                callback      = printMyAsyncCallResult )
```
4) Asynchronous call with invalid input arguments; success
```python
    result = myClient.beginCall(objectAddress = parent,
                                methodAddress = method,
                                inputArgs     = [],
                                callback      = printMyAsyncCallResult )
```

**Test reproduction**
The SampleConsoleServer is provided by default in the Prosys Java OPC-UA SDK evaluation version, which can be downloaded [here](https://www.prosysopc.com/products/opc-ua-java-sdk/). If you want, I can also provide you with a .jar file with the SampleConsoleServer and all the necessary libraries included.

The version of PyUAF that I'm using is from [this commit](https://github.com/uaf/uaf/commit/ef2f219315f41c3ba2f87e45e30286449e4badf3).

**Conclusion**
The asynchronous method calls always succeed, while for the same method, the synchronous calls fail and causes PyUAF to crash. The debug log indicates the problem is located when "Copying the data from SDK level to the UAF result". Calling the same methods with UA Expert works, i.e. the test scenarios can be performed without failures on UA Expert.

 
Do you have any ideas on what the problem might be?

Regards,
Dave

WimPessemier, 2015-06-01:
Hi Dave,

Hmm that shouldn't be happening indeed. Thanks for the detailed report.

I've quickly tried to reproduce a crash using the UaServerCpp demo server, but behaviour seems to be as expected. So I cannot crash PyUAF using the following code: 

```python
import pyuaf
from pyuaf.util                   import Address
from pyuaf.util                   import NodeId
from pyuaf.util                   import RelativePathElement, QualifiedName
from pyuaf.util.primitives        import *
from pyuaf.client                 import Client
from pyuaf.client.settings        import ClientSettings

# list demo server details
DEMO_IP      = "opc.tcp://localhost:48010"
DEMO_NS_URI  = "http://www.unifiedautomation.com/DemoServer"
DEMO_SVR_URI = "urn:UnifiedAutomation:UaServerCpp"

# create the client
myClient = Client(ClientSettings("MyClient", [DEMO_IP]))

# create nodes
demo     = Address(NodeId("Demo", DEMO_NS_URI), DEMO_SVR_URI )
method   = Address(demo,   [ RelativePathElement(QualifiedName("Method"  , DEMO_NS_URI)) ] )
multiply = Address(method, [ RelativePathElement(QualifiedName("Multiply", DEMO_NS_URI)) ] )

# call method.multiply(...)
print myClient.call(method, multiply, [Double(2), Double(4)])            # --> OK, result = Double(8)
print myClient.call(method, multiply, [])                                # --> BadInvalidArgument, as expected
print myClient.call(method, multiply, [Double(2), Double(4), Double(2)]) # --> BadInvalidArgument, as expected
print myClient.call(method, multiply, [Double(2)])                       # --> BadInvalidArgument, as expected
print myClient.call(method, multiply, ["a", Double(4)])                  # --> BadOutOfRange, as expected
```

I'll download and try to run the Prosys demo server, but since I don't know this server, could you please provide me a full example (like the one above) that results in a crash?

Thanks, and I'll keep you posted...

Wim

daveneijsen, 2015-06-01:
Thanks for the quick response!

Below is the example code that causes a crash of PyUAF when run against the SampleConsoleServer of the Prosys OPC UA Java SDK:
```
from pyuaf.client import Client
from pyuaf.client.settings import ClientSettings

from pyuaf.util import *
from pyuaf.util.primitives import *
from pyuaf.util.opcuaidentifiers import *
from pyuaf.util.constants import *
from pyuaf.util.errors import *

# OPC-UA connection constants
ClientName = "myclient"
ServerDiscoveryUrl = "opc.tcp://localhost:48010"
NamespaceUri = "http://www.prosysopc.com/OPCUA/SampleAddressSpace"
ServerUri = "urn:prosysopc.com:OPCUA:SampleConsoleServer"

# Create client with debug logging
settings = ClientSettings()
settings.applicationName = ClientName
settings.discoveryUrls.append(ServerDiscoveryUrl)
settings.logToStdOutLevel = loglevels.Debug
meynClient = Client(settings)

# Set node addresses
parent = Address(NodeId("MyDevice", NamespaceUri), ServerUri)
method = Address(NodeId("MyMethod", NamespaceUri), ServerUri)

# Do the method calls
result = meynClient.call(parent, method, inputArgs=["sin",Double(0)]) # OK
result = meynClient.call(parent, method, inputArgs=[]) # CRASH
```

WimPessemier, 2015-06-01:
I've found the bug that seemed to be causing the error. When you call a method on the Prosys server with invalid arguments, the client SDK sets some attribute called  NoOfOutputArguments to -1. I used this number to resize a vector (a sort of resizable container), but obviously you cannot resize some container to a negative size, hence it failed. The UAF now correctly handles this.

You can also now check the statuscodes for your input arguments more easily, as reported by the server. See the [docs](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client_results.html#pyuaf.client.results.MethodCallResultTarget.inputArgumentOpcUaStatusCodes).

I'm not sure if this will solve all your failing scenarios, as you reported that "Synchronous call with valid input arguments" also fails. I don't have an explanation for this at the moment, if it still fails let me know!

HTH,
Wim

daveneijsen, 2015-06-02:
I've just re-tested the scenarios with your patch and they all pass :)
Thanks for the quick response and fix!

-Dave

closed 2015-06-02 by daveneijsen