uaf/uaf #66: Segmentation fault when using Prosys OPC UA Java SDK

daveneijsen, 2015-06-02:
Hi Wim,

I found a problem when doing certain operations on an OPC UA server created with the Prosys SDK. It seems to be related to the Browse functionality.

The code snippet below reproduces the problem when ran against the Prosys SampleConsoleServer application. It was tested with [this](https://github.com/uaf/uaf/commit/0ea67f5480a7b1de9fe2082e3b9e13feccb27854) version of PyUAF.
```python
from pyuaf.client import Client
from pyuaf.client.settings import ClientSettings

from pyuaf.util import *
from pyuaf.util.primitives import *
from pyuaf.util.opcuaidentifiers import *
from pyuaf.util.constants import *
from pyuaf.util.errors import *

def findNodeByName(foundNodeIds, nodeName, currentBrowseDepth=0,
				   maxBrowseDepth=5, startNodes=[],
				   startNode=opcuaidentifiers.OpcUaId_RootFolder,
				   startNsId=OPCUA_NAMESPACE_ID):
	"""
	:param foundNodeIds: a list with NodeIds of the found nodes (can only contain 0 or 1 items)
	:param nodeName: the displayName of the Node we are looking for
	:param currentBrowseDepth: the recursion level we are in now
	:param maxBrowseDepth: the maximum recursion level we allow
	:param startNodes: a list with nodes where we start browsing on this recursion level
	:param startNode: a single node where we start browsing, specified as an integer value indicating the identifier
					  value of the NodeId, which is converted to a NodeId object later on.
	:param startNsId: the namespaceId that belongs to the startNode
	"""
	if len(foundNodeIds) > 0:
		return
	currentBrowseDepth += 1
	# Check if the browse depth exceeds the maximum to avoid infinite browsing due to circular references
	if maxBrowseDepth != 0 and currentBrowseDepth >= maxBrowseDepth:
		return
	if not startNodes:
		startNodes = [Address(NodeId(startNode, startNsId), ServerUri)]
	browseResult = myClient.browse(startNodes)
	noOfFoundTargets = len(browseResult.targets)
	newNodesToBrowse = []
	for j in xrange(noOfFoundTargets):
		noOfValidReferences = 0
		noOfFoundReferences = len(browseResult.targets[j].references)
		# Each target node to browse has zero or more references
		for i in xrange(noOfFoundReferences):
			if browseResult.targets[j].references[i].displayName.text() == nodeName:
				foundNodeId = browseResult.targets[j].references[i].nodeId.nodeId()
				if foundNodeId not in foundNodeIds:
					foundNodeIds.append(foundNodeId)
					return
			noOfValidReferences += 1
			newNodesToBrowse.append(Address(browseResult.targets[j].references[i].nodeId))
		if noOfValidReferences == 0:
			return
		findNodeByName(foundNodeIds, nodeName, currentBrowseDepth, maxBrowseDepth, newNodesToBrowse)

# OPC-UA connection constants
ClientName = "myClient"
ServerDiscoveryUrl = "opc.tcp://localhost:48010"
NamespaceUri = "http://www.prosysopc.com/OPCUA/SampleAddressSpace"
ServerUri = "urn:prosysopc.com:OPCUA:SampleConsoleServer"

# Create client
settings = ClientSettings()
settings.applicationName = ClientName
settings.discoveryUrls.append(ServerDiscoveryUrl)
#settings.logToStdOutLevel = loglevels.Debug
myClient = Client(settings)

nodeList=[]
findNodeByName(nodeList, "ServerType", startNode=opcuaidentifiers.OpcUaId_BaseObjectType)
print nodeList # This causes segfault
```

It might be related to the [other issue](https://github.com/uaf/uaf/issues/62) with Prosys SDK. Maybe the Prosys SDK returns -1 in more cases, which can cause errors.

Is there any way we can help you in debugging this problem?

Regards,
Dave

WimPessemier, 2015-06-02:
Hi Dave,

Indeed, there were actually 2 problems. They should be fixed now, just pull the latest version!
 
Long explanation in case you're interested:

1. the nodeId() method of ExpandedNodeId returned a C++ *reference*, and therefore SWIG thinks that Python doesn't need to store a copy of this NodeId *object* in memory. Therefore, as Python garbage collected the browseResult when the ``findNodeByName`` function finished, also your NodeId object was removed from memory, which resulted in memory violations when you tried to access it (in order to print it). In principle, this could be avoided in Python ( by using the copy constructor of NodeId: ``foundNodeId = NodeId(ref.nodeId.nodeId())``. But of course it's better to avoid this kind of "black magic", so in the latest version, nodeId() returns a proper C++ object and not a C++ reference anymore, which is handled perfectly well by SWIG.

2. some information was not copied to the nodeId() attribute, this is now also solved.

Just one last thing to note (unrelated to your problem): the PyUAF vectors can really be treated like Python lists, including iterator support. So if you want, you can avoid doing ``for i in xrange(len(...))`` but instead do:
```python
    # ...
    for target in browseResult.targets:
        for ref in target.references:
            if ref.displayName.text() == nodeName:
                foundNodeId = ref.nodeId.nodeId()
                # ...
```

HTH,
Wim

daveneijsen, 2015-06-02:
Thanks again for the quick fix and explanation! It works as expected now.

Regards,
Dave

closed 2015-06-02 by daveneijsen