uaf/uaf #74: Having problems with too many client

Atanamo, 2015-06-08:
Hi Wim,

the following is not a problem with PyUAF, I think... But maybe you have an advice...

We have a gevent-server, which connects to an UA server using the PyUAF library.
This server has a big number of PyUAF client instances - about 40 to 50 (existing simultaneously).
So far we had no problems with it. But from time to time we add new clients… And recently it seemed to hit a limit:

When we surpass a magic number of 45 clients, we encounter strange session problems.
The higher the number of clients that exceed this limit, the more clients lose their session. Consequently, any UA calls on these clients are failing.

Well, we already searched the web and found out, that there seems to be a server-side session or connection limit:
http://forum.unified-automation.com/topic329.html

So the question raises: How to do it correctly? :-)
For now, we plan to refactor our clients in a way, that they do not inherit from the PyUAF Client class any longer.
Instead they could act as a kind of namespace. These namespaces would simply redirect the UA calls to a singleton PyUAF Client then.
(By this, the namespaces would also act as proxies).

But I’m not sure, if it’s a good idea to handle the UA communication of about 50 clients over one single client.

Maybe you have a best practice or any advice for this problem?

Thank you!  ;-)

WimPessemier, 2015-06-09:
Hi,

Hmm it's hard to tell without knowing your application requirements, but interesting in any case :)

I'd first be looking at optimizing the data transfer for a single client, before resorting to more "exotic" solutions with so many clients. Perhaps it's indeed feasible to share a single client among the different parts of your application, and use "custom" callbacks instead of the inherited ones? Or create a sort-of shared memory / database which is filled by OPC UA monitored items and can be accessed by the different parts of your application without the network overhead?

About the optimization: I guess for typical HMI applications there's lot's data flowing from the server to the client, and just a little data/commands going from client to server. The first part (from server to client) can be made extremely efficient by some well chosen subscriptions and monitored items. Many optimizations are possible, e.g.

 - group many monitored items into few subscriptions, and limit the publishing interval of those subscriptions. E.g. if you want to show 100 variables in real-time with an update frequency of 0.1s, add them all to a single subscription with a publishing interval of 0.1s.

 - use DataChangeFilters to avoid sending irrelevant data. E.g. if you're displaying some temperature value to 0.01 degC precision, add a DataChangeFilter to 0.01 degC to let the server filter out the noise.

 - buffer data on the server side. E.g. if you want to display a graph at a sampling rate of 0.01s, set the sampling interval to 0.01s, the buffer size to 10, and add it to the subscription with publishing interval of 0.1s.

 - only group values that change "on command" into a subscription with low publishing interval. E.g., you can easily monitor 1000 configuration variables with a subscription of 30Hz. If the user changes one variable (e.g. by pushing a "+" button which increments a value via the OPC UA Write service), the HMI will be notified 0.03s later and will appear to be very responsive. There's little overhead as a user only changes 1 value at a time.

 - change the monitoring mode of monitored items (or publishing mode of subscriptions) to temporarily disable data transfer, e.g. if a particular tab of the HMI is not visible.

 - for synchronous communication (call/write/...) group as many targets into a single requests. E.g. write an initial configuration of 1000 variables all at once.

If a single shared client is not fast enough, maybe you can share a pool of clients? Something like the example below. It started as a small example but it turned out to become a big one, sorry ;-). It is still not fail-safe: if more than a few 1000 asynchronous requests are queued, 1 result seems to "disappear". I don't have time to fully debug it right now but I do want to develop this example a bit further, as it's a good stress test. I'll also document it better, but you can see there's a queue filled with pyuaf Requests, and this queue is emptied by 10 concurrently running clients. 

So I need to play with it a bit more, but in any case it seems that with 10 clients competing for the same queue, the performance should be significantly better. Also the number of threads are kept under control so it should be more stable too (although not right now as there's some bug in the example :@).

```python
# Start the demo server ($SDK/bin/uaservercpp) of Unified Automation before running this script!

import pyuaf
from pyuaf.util             import Address, NodeId
from pyuaf.client           import Client
from pyuaf.client.settings  import ClientSettings
from pyuaf.client.requests  import ReadRequest, AsyncReadRequest
import time
import thread, threading, Queue
from datetime import datetime

# specify the address of the node of which we would like to read its Value attribute:
# (here we're using an absolute address, i.e. a NodeId(<identifier>, <namespace URI>) and a <server URI>)
someAddress = Address(NodeId("Demo.SimulationSpeed",                         # NodeId identifier
                             "http://www.unifiedautomation.com/DemoServer"), # NodeId namespace URI
                      "urn:UnifiedAutomation:UaServerCpp")                   # server URI

class Pool(threading.Thread):
    
    def __init__(self, noOfWorkers=10):
        self.queue       = Queue.Queue()
        self.noOfWorkers = noOfWorkers
        self.clients     = {} # number : client
        self.settings    = ClientSettings("myClient", ["opc.tcp://localhost:48010"])
        #self.settings.logToStdOutLevel = pyuaf.util.loglevels.Debug
        self.results     = {} # key : result
        self.uniqueKey   = 0
        self.lock        = threading.Lock()
        self.loggingLock = threading.Lock()
        
        for i in xrange(noOfWorkers):
            thread.start_new_thread(self.addWorker, (i,))
    
    def log(self, msg):
        try:
            self.loggingLock.acquire()
            print("%s  %s" %(datetime.now().strftime("%H:%M:%S.%f"), msg))
        finally:
            self.loggingLock.release()

    def shutdown(self):
        for i in xrange(self.noOfWorkers):
            self.putRequest(None)
        
        # wait until the clients are destructed
[... 179 lines cut ...]
    t.startSyncTests()
    t.checkSyncTests()
    syncTestEndTime = time.time()
    
    print("ASYNC TEST:")
    asyncTestStartTime = time.time()
    t.startAsyncTests()
    t.checkAsyncTests()
    asyncTestEndTime = time.time()
[... cut ...]
    print("SYNC TEST  : Time needed for %d tests with %d workers: %.3f seconds" %(NO_OF_TESTS, NO_OF_WORKERS, syncTestEndTime - syncTestStartTime))
    print("ASYNC TEST : Time needed for %d tests with %d workers: %.3f seconds" %(NO_OF_TESTS, NO_OF_WORKERS, asyncTestEndTime - asyncTestStartTime))

finally:
    t.shutdown() # important, otherwise the threads don't exit cleanly and the 
                 # clients are not destructed properly. And if they clients are not 
                 # destructed properly, the sessions remain active, and the server will 
                 # soon start to complain about TooManySessions!

```

HTH,

Wim

Atanamo, 2015-06-09:
Wow, I didn't expect a big answer like this :-D
Thanks very much!

The concept of the client pool, especially the request queue, is very interesting. 
But I'm not sure, if we need real multithreading yet: Our gevent server does a good job so far, though it's basically a single threaded server (But there are some more threads, created by the webserver itself etc...). Therefor by our current concept of many clients all data is piped to a single thread - which makes the multiple client threads kinda needless, I think.
Nevertheless we may have to switch to multithreading in future (it's a steadily evolving app ^^). So your example is an excellent one to do so :-)

But for the first step we will try to share a single UA client then. I think, we can optimize some things as you suggested, too :-)
Especially temporarily deactivation of monitored items and the DataChangeFilters sounds promising.
When we have new insights from this approach, I let you know ;-)

Best regards and thanks again! :-)