uaf/uaf #76: browseName uses lower case namespace

jorrita, 2015-06-15:
This may be intentional, but when I browse a node, its browseName seems always lower case. Also, a traling slash seems to go missing :)

```
>>> objectNodeAddress = Address(NodeId(OpcUaId_ObjectsFolder, OPCUA_NAMESPACE_ID), "urn:UnifiedAutomation:UaServerCpp")
>>> ServerNode = QualifiedName('Server', OPCUA_NAMESPACE_URI, OPCUA_NAMESPACE_ID)
>>> res = cppdemoClient.browse([objectNodeAddress])
>>> res.targets[0].references[1].browseName
<pyuaf.util.QualifiedName(Ns=0='http://opcfoundation.org/ua'|'Server')>
>>> ServerNode
<pyuaf.util.QualifiedName(Ns=0='http://opcfoundation.org/UA/'|'Server')>
```

WimPessemier, 2015-06-15:
Hmm yes this is a relic of the first versions of the framework... At that point I thought it would be better to normalize the namespace URIs internally, so as to make comparison more "tolerant". So indeed, trailing slashes are removed and URIs are made lowercase.
I guess this normalization is not according to the specs, although I should look that up first. Also it may not be performed everywhere throughout the whole framework (as you noted).

I'll check the specs and remove the normalization code if needed (unless someone sees good reasons to keep it).

jorrita, 2015-06-15:
I see - no problem: the reason that I wanted to use the QualifiedNames in the first place was a little silly anyway. I wanted to bail out of browsing when encountering the Server node, but it is much simpler to compare the current nodeId to pyuaf.util.opcuaidentifiers.OpcUaId_Server