uaf/uaf #77: OPC UA status code

daveneijsen, 2015-06-23:
Hi Wim,

I'm trying to get the OPC UA status code (defined in pyuaf.util.opcuastatuscodes) from a Status object. Currently I'm only able to get the UAF status code, but the documentation states that the Status object also includes the OPC UA status code. Do you know how I can retrieve the OPC UA status code?
Regards,
Dave

WimPessemier, 2015-06-23:
Hi Dave,

Which UAF version are you using? 

- In the latest official release (v1.10) you could get the OPC UA status code from the Status object:
```python
print(myStatus.opcUaStatusCode())
```
 
- In the current master branch (version pre-2.0.0), the OPC UA status is either given via the Exception that was raised, or via the result directly.
For example:
```python
try:
   result = myClient.read([myAddress])
   print(result.targets[0].opcUaStatusCode)
except pyuaf.util.errors.ReadInvocationError, e:
   print(e.sdkStatus.statusCode)
except pyuaf.util.errors.UafError, e:
   print("oops, some other error")
   print(e)  
```
See 
 - http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_util.html#class-sdkstatus
 - http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_util_errors.html#class-uaferror

If some statuscode is missing or there's some other problem, let me know!
Wim

daveneijsen, 2015-06-23:
`result.targets[0].opcUaStatusCode` is exactly what I need and it works perfectly when using it to get the OPC UA status code from a method call. However, the documentation doesn't include the `opcUaStatusCode` property yet I think: http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client_results.html#class-methodcallresulttarget.

Anyway, it works, so closing the issue :)

closed 2015-06-23 by daveneijsen

WimPessemier, 2015-06-23:
Okay thanks for the notice, it's indeed missing in the docs. I will make sure they get updated :)