uaf/uaf #78: Accessing GUID value in Variable Node

daveneijsen, 2015-06-24:
Hi Wim,

Today I'm trying to read  GUID value from a Node of type Variable. In the example below you can see that the actual data is in the ReadResultTarget, but I'm getting errors when trying to access it:
```
>>> r.targets[0]
<pyuaf.client.results.ReadResultTarget( - clientConnectionId : 0
                                        - status             : Good
                                        - opcUaStatusCode    : 0
                                        - data               : {918755c7-e52c-4180-8bd6-5a68f5300062}
                                        - sourceTimestamp    : 2015-06-24T11:19:57.972Z
                                        - serverTimestamp    : 2015-06-24T11:24:01.688Z
                                        - sourcePicoseconds  : 0
                                        - serverPicoseconds  : 0)>

>>> r.targets[0].data
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Unsupported type!

>>> dir(r.targets[0].data)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Unsupported type!

>>> dir(r.targets[0].data.value)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Unsupported type!
```
For some reason I cannot access the data. Do you have any ideas?

-Dave

WimPessemier, 2015-06-24:
Hi Dave, indeed I apparently forgot to include GUIDs in a SWIG typemap (which automatically converts any data into the correct data type). 
Fixing should be easy, I'll try to do it soon.
I'll keep you posted,
Wim

WimPessemier, 2015-06-26:
Preliminary solution here:
https://github.com/uaf/uaf/compare/master...WimPessemier:master
When it's tested on all platforms, I'll create a pull request to the UAF:master branch.

WimPessemier, 2015-07-17:
This bug was fixed in v2.0.0.

closed 2015-07-17 by WimPessemier