uaf/uaf #79: LocalizedText does strange things

jorrita, 2015-06-25:
It looks a bit like unicode strings in LocalizedText have a problem. In the below example (using uaservercpp) we get random-looking values back into our dict (same behaviour if using a list)
Note that we are using python2.7 and it is quite possible that this is not a problem in pyuaf, but in Python2 itself.  

```
import pyuaf
from pyuaf.client           import Client
from pyuaf.client.settings  import ClientSettings
from pyuaf.util             import Address, NodeId, ExpandedNodeId

cppnameSpace = "http://www.unifiedautomation.com/DemoServer"
cppdemoUri = "urn:UnifiedAutomation:UaServerCpp"
cppdemosettings = ClientSettings("cppdemoclient", ["opc.tcp://localhost:48011"])
client = Client(cppdemosettings)
a = Address(ExpandedNodeId("Demo.UnicodeTest", cppnameSpace, cppdemoUri))

for t in range(100):
  h={}
  res=client.browse([a])
  idx=0
  for r in res.targets[0].references:
    if r.referenceTypeId.identifier().idNumeric == 35:
      print idx, r.displayName
      h[idx] = r.displayName
      idx+=1
  print h
```

WimPessemier, 2015-06-25:
Hi,
hmm yes this is one of the dark corners of SWIG (or perhaps, of my understanding of SWIG).
The problem is concerned with SWIG vs. Python ownership (Python's INCREF and DECREF). SWIG apparently does not keep ownership of your reference (``r``), so by the time you do ``print h``, it points to invalid memory.
 
Two ways to circumvent the problem:
- use the copy constructor of LocalizedText to **explicitly** store your LocalizedText in memory:
   - replace line ``h[idx] = r.displayName``
   - with ``h[idx] = LocalizedText(r.displayName)``
- or, don't use the iterator of the ``res.targets[0].references`` wrapped vector, but access the vector items indirectly:
  - replace line ``for r in res.targets[0].references:``
  - with lines 
```python
for i in xrange(len(res.targets[0].references)):
   r = res.targets[0].references[i]  # SWIG will store the reference in memory now
```

Both approaches appear to be a bit "black magic", but it's the best answer I can give so far...

HTH,
Wim

jorrita, 2015-06-25:
OK, we can live with that (the copy constructor seems the least invasive approach). 
Do you want to close this issue or keep it open until a more fundamental resolution is found?

closed 2015-07-10 by jorrita