uaf/uaf #80: read() on a no-existing address gives overallStatus(Good)

jorrita, 2015-07-14:
In the following example, I would expect the overallStatus of a read() to be Bad instead of Good...
```
>>> e = pyuaf.util.ExpandedNodeId("aaapnootmies", NamespaceUri. ServerUri)
>>> a = pyuaf.util.Address(e)
>>> res = client.read(a)
>>> res.overallStatus
<pyuaf.util.Status(Good)>
>>> res = client.browse(a)
>>> res.overallStatus
<pyuaf.util.Status(Bad statuses are present)>

WimPessemier, 2015-07-14:
You're right, here's a quick patch if you need one:
```
diff --git a/src/uaf/client/invocations/readinvocation.cpp b/src/uaf/client/invocations/readinvocation.cpp
index c955778..5d3e69e 100644
--- a/src/uaf/client/invocations/readinvocation.cpp
+++ b/src/uaf/client/invocations/readinvocation.cpp
@@ -164,6 +164,12 @@ namespace uaf
                 targets[i].fromSdk(UaDataValue(uaDataValues_[i]));
                 nameSpaceArray.fillVariant(targets[i].data);
                 serverArray.fillVariant(targets[i].data);
+
+                // update the status
+                if (OpcUa_IsGood(uaDataValues_[i].StatusCode))
+                    targets[i].status = statuscodes::Good;
+                else
+                    targets[i].status = ServerCouldNotWriteError(SdkStatus(uaDataValues_[i].StatusCode));
             }
 
             ret = statuscodes::Good;
```

I'm about to release v2.0.0 so I will include the bugfix in that version.

closed 2015-07-14 by WimPessemier

jorrita, 2015-07-15:
Dank!

Closed #80<https://github.com/uaf/uaf/issues/80>.