uaf/uaf #81: Construct an ExpandedNodeId with a Guid

jorrita, 2015-07-15:
In our software, we use Guids as node identifiers. In pyuaf, it is possible to construct a NodeId with a Guid, but not an ExpandedNodeId. Is that intentional?

WimPessemier, 2015-07-15:
Hi,
You can always create ExpandedNodeIds based on a Guid-NodeId like this:
```python
ExpandedNodeId(NodeId(Guid(), "nsUri"), "svrUri")
```
It's true that there are some "shortcuts" to frequently used constructions, e.g.
```python
assert ExpandedNodeId(NodeId("stringid", "nsUri"), "svrUri") == ExpandedNodeId("stringid", "nsUri", "svrUri")
assert ExpandedNodeId(NodeId(123, "nsUri"), "svrUri") == ExpandedNodeId(123, "nsUri", "svrUri")
```
If you want, I can also add a shortcut (= extra constructor for ExpandedNodeId) so that:
```python
assert ExpandedNodeId(NodeId(Guid(), "nsUri"), "svrUri") == ExpandedNodeId(Guid(), "nsUri", "svrUri") # NOT POSSIBLE (YET)
```
I hope this answers your question, let me know if it doesn't. ;)

jorrita, 2015-07-15:
Sure! 
I originally wrote this as a feature request, but in the end I decided that it was not that important. I just like things to be symmetrical is all :)
So if you feel it adds enough value to the library, please add the extra constructor. If not, I'll just go keep on going through NodeId(Guid(guid),nsUri)

closed 2015-07-15 by jorrita

WimPessemier, 2015-07-16:
I'll see if I can find some time to add the constructors. But it's just 2 files that need to be extended a bit (src/uaf/util/expandednodeid.h and src/uaf/util/expandednodeid.cpp), it's really simple, if you make a pull request I'll happily merge it.