uaf/uaf #85: Segv in AddressCache

jorrita, 2015-08-04:
When running our automated tests on our OPC UA server, pyuaf gives a SEGV.

This seems to happen when creating multiple Client instances with different localeId's
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fbd35612da2 in uaf::operator==(uaf::NodeIdIdentifier const&, uaf::NodeIdIdentifier const&) ()
   from /usr/lib/python2.7/site-packages/libuafutil.so
Missing separate debuginfos, use: debuginfo-install opcua-sdk-libs-1.2-1.fc21.x86_64
(gdb) where
#0  0x00007fbd35612da2 in uaf::operator==(uaf::NodeIdIdentifier const&, uaf::NodeIdIdentifier const&) ()
   from /usr/lib/python2.7/site-packages/libuafutil.so
#1  0x00007fbd3565d829 in uaf::operator==(uaf::NodeId const&, uaf::NodeId const&) ()
   from /usr/lib/python2.7/site-packages/libuafutil.so
#2  0x00007fbd3565d869 in uaf::operator!=(uaf::NodeId const&, uaf::NodeId const&) ()
   from /usr/lib/python2.7/site-packages/libuafutil.so
#3  0x00007fbd3566868f in uaf::operator<(uaf::ExpandedNodeId const&, uaf::ExpandedNodeId const&) ()
   from /usr/lib/python2.7/site-packages/libuafutil.so
#4  0x00007fbd30445b00 in uaf::AddressCache::find(uaf::Address const&, uaf::ExpandedNodeId&) ()
   from /usr/lib/python2.7/site-packages/libuafclient.so
#5  0x00007fbd3049cd9f in uaf::Resolver::resolveCached(std::vector<uaf::Address, std::allocator<uaf::Address> > const&, std::vector<uaf::ExpandedNodeId, std::allocator<uaf::ExpandedNodeId> >&, std::vector<uaf::Status, std::allocator<uaf::Status> >&, uaf::Mask&, uaf::Mask&) () from /usr/lib/python2.7/site-packages/libuafclient.so
#6  0x00007fbd3049e469 in uaf::Resolver::resolve(std::vector<uaf::Address, std::allocator<uaf::Address> > const&, std::vector<uaf::ExpandedNodeId, std::allocator<uaf::ExpandedNodeId> >&, std::vector<uaf::Status, std::allocator<uaf::Status> >&) () from /usr/lib/python2.7/site-packages/libuafclient.so
#7  0x00007fbd304213df in uaf::Status uaf::Resolver::resolve<uaf::MethodCallService>(uaf::MethodCallService::Request&, uaf::Mask const&, uaf::MethodCallService::Result&) () from /usr/lib/python2.7/site-packages/libuafclient.so
#8  0x00007fbd3043bc9a in uaf::Status uaf::Client::processRequest<uaf::MethodCallService>(uaf::MethodCallService::Request const&, uaf::Mask const&, uaf::MethodCallService::Result&) () from /usr/lib/python2.7/site-packages/libuafclient.so
#9  0x00007fbd303bf4af in uaf::Client::processRequest(uaf::BaseSessionRequest<uaf::MethodCallSettings, uaf::MethodCallRequestTarget, false> const&, uaf::BaseSessionResult<uaf::MethodCallResultTarget, false>&) ()
   from /usr/lib/python2.7/site-packages/libuafclient.so
#10 0x00007fbd303bf826 in uaf::Client::call(uaf::Address const&, uaf::Address const&, std::vector<uaf::Variant, std::allocator<uaf::Variant> > const&, unsigned int, uaf::MethodCallSettings const*, uaf::TranslateBrowsePathsToNodeIdsSettings const*, uaf::SessionSettings const*, uaf::BaseSessionResult<uaf::MethodCallResultTarget, false>&) ()
   from /usr/lib/python2.7/site-packages/libuafclient.so
#11 0x00007fbd307f4888 in _wrap_ClientBase_call () from /usr/lib/python2.7/site-packages/pyuaf/client/___init__.so
#12 0x00007fbd42d71a5e in call_function (oparg=<optimized out>, pp_stack=0x7ffe65c92fc0)
    at /usr/src/debug/Python-2.7.8/Python/ceval.c:4110
[...]

WimPessemier, 2015-08-05:
Hi, just to let you know that I've experimented a bit with the code you sent me (I'm able to run the server and the tests), but haven't gotten to the bottom of it. It seems that the Client tries to access cached addresses which point to invalid memory. It may be related to SWIG freeing up memory when it shouldn't, but as said, I need some more time to investigate it.
I'll keep you posted.
Wim

jorrita, 2015-08-06:
Thanks!
Voor de goede orde: bij wijze van work-around heb ik de code wat aangepast; per testcase gebruiken we nu steeds maar één Client waarvan we steeds de settings.localeId veranderen (is ook handiger eigenlijk).
Voor mij is het vooral belangrijk te weten wat er nu aan de hand is, zodat ik er minimaal omheen kan blijven werken…
Jorrit

WimPessemier, 2015-08-06:
Hi,
I've experimented a bit more with your test code, and although it's much too big to easily debug, I think the problem is related to Addresses being deleted when they are still needed.

Consider the following:
```python
rootAddress = Address(...)
someRelativePath = [ RelativePathElement(...) ]
myAddress = Address(rootAddress, someRelativePath)
```

``myAddress`` doesn't keep a full copy of ``rootAddress`` in memory, it only stores a pointer. So if ``rootAddress`` gets deleted for some reason, the UAF will segfault when it tries to access ``myAddress``. Since ``myAddress`` is automatically cached by the framework, it may "hang around" for the whole lifetime of the Client. 

Anyway it seems that if I change the code so that ``myAddress`` keeps a copy of ``rootAddress``, all of your tests pass without segfaults. Could you please try the patch below and test it a bit further?

The modified code is a bit less memory-efficient, especially for "deep" hierarchies of addresses (as each "child address" will store a full copy of its parent address, which will contain again a copy of its parent address, and so on). But in practice I guess the extra memory consumption will be irrelevant, so if the patch below works for you, then I will create version 2.0.1 out of it.

HTH,
Wim

```diff
diff --git a/src/uaf/util/address.cpp b/src/uaf/util/address.cpp
index 5f860dd..137406c 100644
--- a/src/uaf/util/address.cpp
+++ b/src/uaf/util/address.cpp
@@ -45,9 +45,9 @@ namespace uaf
     // =============================================================================================
     Address::Address(Address* startingAddress, const vector<RelativePathElement>& relativePath)
     : isRelativePath_(true),
-      isStartingAddressOwned_(false),
+      isStartingAddressOwned_(true),
       relativePath_(new vector<RelativePathElement>(relativePath)),
-      startingAddress_(startingAddress),
+      startingAddress_(new Address(*startingAddress)),
       expandedNodeId_(0)
     {}
 
@@ -56,8 +56,8 @@ namespace uaf
     // =============================================================================================
     Address::Address(Address* startingAddress, const RelativePathElement& relativePath)
     : isRelativePath_(true),
-      isStartingAddressOwned_(false),
-      startingAddress_(startingAddress),
+      isStartingAddressOwned_(true),
+      startingAddress_(new Address(*startingAddress)),
       expandedNodeId_(0)
     {
         relativePath_ = new vector<RelativePathElement>();
```

jorrita, 2015-08-06:
This patch solves the crash that we experienced.
Thank you!
Jorrit

closed 2015-08-06 by jorrita

WimPessemier, 2015-08-21:
This bug was fixed in v2.0.1.