uaf/uaf #86: Memort leak in Client::write and Client::call

jorrita, 2015-08-17:
It looks like pyuaf leaks memory.

(examples use uaservercpp running on port 48011)
example 1: writing existing values to an existing Address
```
>>> import pyuaf
>>> from pyuaf.client import Client
>>> from pyuaf.client.settings import ClientSettings
>>> client=Client(ClientSettings("client", ["opc.tcp://localhost:48011"]))
>>> ad = Address(NodeId("Demo.BoilerDemo.Boiler1.HeaterStatus", 2), "urn:UnifiedAutomation:UaServerCpp")
>>> val0 = pyuaf.util.primitives.Int32(0)
>>> val1 = pyuaf.util.primitives.Int32(1)
>>> r=os.system("ps aux|grep python|grep -v grep")
jorrita  31578  0.2  1.0 603028 42648 pts/4    Sl+  06:51   0:05 python
>>> for i in range(100):
...   r=client.write([ad], [val0])
...   r=client.write([ad], [val1])
... 
>>> r=os.system("ps aux|grep python|grep -v grep")
jorrita  31578  0.2  1.1 605944 45444 pts/4    Sl+  06:51   0:06 python
>>> 
```

example 2: calling a method 
```
>>> parentad = Address(NodeId("Demo.DynamicNodes", 2), "urn:UnifiedAutomation:UaServerCpp")
>>> methodad1 = Address(NodeId("Demo.DynamicNodes.CreateDynamicNode", 2), "urn:UnifiedAutomation:UaServerCpp")
>>> methodad2 = Address(NodeId("Demo.DynamicNodes.DeleteDynamicNode", 2), "urn:UnifiedAutomation:UaServerCpp")      
>>> r=os.system("ps aux|grep python|grep -v grep")
jorrita  31578  0.2  1.1 605944 45588 pts/4    Sl+  06:51   0:07 python
>>> for i in range(100):
...   r=client.call(parentad, methodad1, [])
...   r=client.call(parentad, methodad2, [])
... 
>>> r=os.system("ps aux|grep python|grep -v grep")
jorrita  31578  0.3  1.1 608860 48544 pts/4    Sl+  06:51   0:07 python

```

jorrita, 2015-08-21:
With some help from a collegue, I can supply a fix that seems to solve the biggest problem.
As you can see, this proposal uses c++11, so you also need to change some CMakefiles...

```
diff --git a/src/uaf/client/sessions/sessionfactory.h b/src/uaf/client/sessions/sessionfactory.h
index 36de519..8f66751 100644
--- a/src/uaf/client/sessions/sessionfactory.h
+++ b/src/uaf/client/sessions/sessionfactory.h
@@ -22,6 +22,7 @@
 #define UAF_SESSIONFACTORY_H_
 
 // STD
+#include <memory>
 #include <vector>
 #include <string>
 #include <sstream>
@@ -347,7 +348,7 @@ namespace uaf
             bool handleStored = storeRequestHandleIfNeeded<_Service>(request, transactionId);
 
             // create a map to store the invocations that we'll create
-            typedef std::map<uaf::Session*, Invocation*> InvocationMap;
+            typedef std::map<uaf::Session*, std::unique_ptr<Invocation> > InvocationMap;
             InvocationMap invocations;
 
             logger_->debug("Building the invocations");
@@ -366,7 +367,7 @@ namespace uaf
                             ret = acquireExistingSession(request.clientConnectionId, session);
                             if (ret.isGood())
                             {
-                                invocations[session] = new Invocation;
+                                invocations[session].reset(new Invocation);
                                 invocations[session]->setAsynchronous(async);
                                 invocations[session]->setRequestHandle(requestHandle);
                                 invocations[session]->setServiceSettings(getServiceSettings<_Service>(request));
@@ -424,7 +425,7 @@ namespace uaf
                                 if (ret.isGood())
                                 {
                                     logger_->debug("Scheduling an invocation for this session");
-                                    invocations[session] = new Invocation;
+                                    invocations[session].reset(new Invocation);
                                     invocations[session]->setAsynchronous(async);
                                     invocations[session]->setRequestHandle(requestHandle);
                                     invocations[session]->setServiceSettings(getServiceSettings<_Service>(request));
@@ -469,7 +470,7 @@ namespace uaf
 
                 // create a pointer to the current invocation
                 uaf::Session* session = it->first;
-                Invocation*   invocation = it->second;
+                const std::unique_ptr<Invocation>& invocation = it->second;
 
                 // set the transactionId if necessary
                 if (handleStored)

```

WimPessemier, 2015-08-21:
Hi,
Sorry for not responding sooner but I had a terribly busy week...
Thanks for investing time in this. In the UAF version < 2.0.0 there was an "InvocationFactory" class that took care of deleting the invocations after they were processed. But in v2.0.0. the InvocationFactory became obsolete, and I forgot to delete the invocations in the SessionFactory, as you discovered.

The bug can easily be fixed without C++11, see patch:
```patch
diff --git a/src/uaf/client/sessions/sessionfactory.h b/src/uaf/client/sessions/sessionfactory.h
index 36de519..3012bb4 100644
--- a/src/uaf/client/sessions/sessionfactory.h
+++ b/src/uaf/client/sessions/sessionfactory.h
@@ -38,7 +38,6 @@
 #include "uaf/client/requests/requests.h"
 #include "uaf/client/results/results.h"
 #include "uaf/client/settings/allsettings.h"
-#include "uaf/client/invocations/invocationfactory.h"
 
 
 namespace uaf
@@ -501,9 +500,16 @@ namespace uaf
 
                 releaseSession(session);
 
+                // don't forget to delete the invocation!!!
+                // (see bugfix https://github.com/uaf/uaf/issues/86)
+                delete invocation;
+
                 invocationIndex++;
             }
 
+            // clear the InvocationMap
+            invocations.clear();
+
             // remove the handle if one was stored, and if there was an unexpected error
             if (ret.isNotGood() && handleStored)
             {
```

I'm running the following script based on your example code:
```python
import resource
import pyuaf
from pyuaf.client import Client
from pyuaf.client.settings import ClientSettings
from pyuaf.util import Address, NodeId

def getMemoryUsage():
    return resource.getrusage(resource.RUSAGE_SELF).ru_maxrss

client = Client(ClientSettings("myClient", ["opc.tcp://localhost:48010"]))
ad     = Address(NodeId("Demo.BoilerDemo.Boiler1.HeaterStatus", 2), "urn:UnifiedAutomation:UaServerCpp")
val0   = pyuaf.util.primitives.Int32(0)

# on first invocation a session will be created, resulting in memory being consumed. 
r = client.write([ad], [val0])

for noOfTests in [1,10,100,1000,10000,100000]:
    
    print '%s tests:' %noOfTests
    
    before = getMemoryUsage()
    print ' - before : %s (kb)' %before

    for i in xrange(noOfTests):
        r = client.write([ad], [val0])
    
    after = getMemoryUsage()
    print ' - after  : %s (kb)' %after
    print ' - leaked : %s (kb)' %(after-before)
```

This results in:
```
1 tests:
 - before : 32360 (kb)
 - after  : 32360 (kb)
 - leaked : 0 (kb)
10 tests:
 - before : 32360 (kb)
 - after  : 32360 (kb)
 - leaked : 0 (kb)
100 tests:
 - before : 32360 (kb)
 - after  : 32360 (kb)
 - leaked : 0 (kb)
1000 tests:
 - before : 32360 (kb)
 - after  : 32360 (kb)
 - leaked : 0 (kb)
10000 tests:
 - before : 32360 (kb)
 - after  : 32360 (kb)
 - leaked : 0 (kb)
100000 tests:
 - before : 32360 (kb)
 - after  : 32520 (kb)
 - leaked : 160 (kb)
```

For some reason there is 160 kb of memory leaking when going from 10K to 100K invocations. I've got no explanation for this, but luckily this is very minimal.

In any case it would be useful to distil some unit tests (for read/write/call etc) that check for these kind of memory leaks, because there may be others. I'll put it on the todo list.

I'll release this bugfix and the previous one ASAP.

Thanks again,

Wim

jorrita, 2015-08-24:
Hoi Wim,
Graag gedaan en geen probleem als je al die nieuwlichterij met *_ptr niet wilt gebruiken ☺
We gaan 2.0.1 naar binnen halen; dank voor je snelle integratie in de master.
Jorrit

closed 2015-08-24 by jorrita