uaf/uaf #88: there are no discovery URLs exposed by Labview OPC UA server

gitcyc, 2015-08-25:
I'am using pyuaf-client to connect to Labview OPC UA server (labview\examples\comm\OPCUA), but failed. And the following error messagea are printed:
        
 [<pyuaf.util.ApplicationDescription( - applicationType: 0 (Server)
                                    - applicationName: OPCUA Server
                                    - applicationUri: NationalInstruments.uaserver.LabVIEW
                                    - productUri:
                                    - discoveryUrls[]: []
                                    - discoveryProfileUri:
                                    - gatewayServerUri: )>]
NationalInstruments.uaserver.LabVIEW
 - type             : ExpandedNodeId
 - expandedNodeId   : Svr='NationalInstruments.uaserver.LabVIEW'|Ns='http://opcfoundation.org/UA/'|String|Device.Node
Traceback (most recent call last):
  File "easiest_client_example-4.py", line 74, in <module>
    result_item = myClient.read( [someAddress_item] )
  File "/home/dslab/bd_test/uaf/lib/pyuaf/client/__init__.py", line 1924, in read
    result).test()
  File "/home/dslab/bd_test/uaf/lib/pyuaf/util/__init__.py", line 1003, in test
    raise getattr(self, "raisedBy_%s" %self.statusCodeName())
pyuaf.util.errors.NoDiscoveryUrlsExposedByServerError: Server NationalInstruments.uaserver.LabVIEW has been discovered, but there are no discovery URLs exposed by it

    I don't know what happed for that. So does this mean labview OPC UA doesn't support discovery services? By the way, if we use LDS, pyuaf-client can successfully connect to labview opc ua server.

WimPessemier, 2015-08-25:
Hi,
it seems that your server is not configured correctly as there are no "connectable" endpoints exposed (``discoveryUrls`` is an empty list).
Can you connect to the server using UaExpert? If you browse the discovery server (with url ``opc.tcp://<ipaddress>:<port>`` using UaExport, can you see some endpoints to which you can connect (for example, an unsecured one and an encrypted/signed one)?
PyUAF doesn't require discovery in fact. It is advisable to use discovery, but if you want you can directly connect to the server endpoint. See [this example](https://github.com/uaf/uaf/blob/master/examples/pyuaf/client/how_to_manually_connect_to_an_endpoint_without_discovery.py).
If possible, post some screenshots of UaExpert!

WimPessemier, 2015-08-29:
Hi, did you finally solve the problem with your LabVIEW server? Please comment or re-open this issue if needed, thanks.
Wim

closed 2015-08-29 by WimPessemier

gitcyc, 2015-08-31:
Thanks a lot for your reply. Your method is very effective for connecting LabView OPC UA server.