uaf/uaf #89: idGuid property contains invalid data

daveneijsen, 2015-09-09:
The idGuid property of a NodeIdIdentifier object seems to contain an incorrect value:

```
>>> guid=pyuaf.util.Guid(str(uuid.uuid4()))
>>> guid
<pyuaf.util.Guid({92da3bdd-ff14-4787-8792-3f073beccdb9})>
>>> node=pyuaf.util.NodeId(guid, 1)
>>> node
<pyuaf.util.NodeId(Ns=1|Guid|{92da3bdd-ff14-4787-8792-3f073beccdb9})>
>>> node.identifier()
<pyuaf.util.NodeIdIdentifier(Guid|{92da3bdd-ff14-4787-8792-3f073beccdb9})>
>>> node.identifier().idGuid
<pyuaf.util.Guid({00000000-0000-0000-0000-000000000000})>
>>> node.identifier().idGuid
<pyuaf.util.Guid({00000000-703c-7579-6166-2e7574696c2e})>
>>> node.identifier().idGuid
<pyuaf.util.Guid({00000000-703c-7579-6166-2e7574696c2e})>
```

We have a workaround by extracting the GUID from the string representation (__str__) of the NodeIdIdentifier object, so the issue is not blocking us right now.

Note: when accessing the property multiple times the values changes (as seen in the example). Could be some pointer to random memory or SWIG issue?

WimPessemier, 2015-09-09:
Hmm I had a quick look at the code but I cannot see immediately what's causing the problem.
I suspect there is an internal (private) SDK object which keeps the correct value, but this value is not copied to the ``idGuid`` attribute (which has been created for easy wrapping by SWIG).
As this issue is not blocking you, I will look at it early next week (as this week is very busy due to some  commissioning work abroad).

WimPessemier, 2015-09-14:
I think this is (again) a SWIG issue: 
```python
from pyuaf.util import NodeId, Guid
node = NodeId( Guid('{92da3bdd-ff14-4787-8792-3f073beccdb9}'), 1)
print node.identifier().idGuid # BAD
id = node.identifier()
print id.idGuid # GOOD
```

I'll see if I can find something on the SWIG lists about this issue. If you see something, let me know.