uaf/uaf #90: String constructor can't handle unicode strings

jorrita, 2015-11-09:
I am trying to construct a String with unicode contents (analogous to UaString):
>>> uncodeString=u'\u2030'
>>> s=pyuaf.util.primitives.String(uncodeString)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/pyuaf/util/primitives.py", line 964, in __init__
    raise TypeError("'str' expected instead of %s instance" %type(args[0]))
TypeError: 'str' expected instead of <type 'unicode'> instance

The problem seems to be, that SETUP_PRIMITIVES in util_primitives.i only accepts one type (currently <str>). In python2, <unicode> and <str> are siblings, both deriving from <basestring>.

The question: is this solvable using python2.x, or is the only way to move to python3? (tangentially: Is python3 supported by uaf?)

WimPessemier, 2015-11-09:
Hi,
In principle PyUAF expects UTF-8 encoded strings.
Is this giving you expected results?
```python
uncodeString=u'\u2030'
s=pyuaf.util.primitives.String(uncodeString.encode("UTF-8"))
```

I have never tested PyUAF with Python 3, but SWIG does support it. 
So I guess this means that there may be some tweaking necessary, but I cannot image it would be a big effort to make it work...

jorrita, 2015-11-09:
Hmm, apparently I have misinterpreted the unicode documentation... 
I was under the impression that the unicode string itself is fundamentally different from its encoding, but it looks like they may be used interchangeably...

closed 2015-11-09 by jorrita

WimPessemier, 2015-11-09:
I think basically a ``Str`` is just a container for 8-bit characters (at least, in Python 2.x). 
So if you want to fill a ``Str`` with 16-bit characters (as you're doing), you need to convert (encode) them to a number of 8-bit characters.

jorrita, 2015-11-09:
Ooh don’t get me started on string/unicode handling in python2 ☺
Ik vind een betere manier om erover te denken: ik wil een “Unicode string” (wat dat precies ook betekent) in een pyuaf String stoppen. Maar python2 is een beetje onhandig, dus nu moet ik een unicode string verminken want (pyuaf) String accepteert geen unicode-objecten.

Ik zal nog een keer proberen om een pyuaf RPM te maken met python3, maar tot nu toe liep dat vast… Het lijkt erop dat het niet helemaal triviaal is ten opzichte van het maken van een RPM voor python2.
Jorrit