uaf/uaf #91: pkipublickey: support C++ based OPC UA Client SDK Bundle v1.5.0

gitcyc, 2015-11-17:
Hi Wim,
    I tried to use `C++ based OPC UA Client SDK Bundle v1.5.0` for building  (Py)UAF, but failed.
I think there need some relatively minor modifications for the latest (Py)UAF source code. The 
following patch can work well.

thx!
   cyc

==================================================================
From e69d5dd11edb0c83052e10c553a000a976ca5596 Mon Sep 17 00:00:00 2001
From: Chen Yucong <chenyc@impcas.ac.cn>
Date: Tue, 17 Nov 2015 11:52:43 +0800
Subject: [PATCH] pkipublickey: support C++ based OPC UA Client SDK Bundle
 v1.5.0

---
 src/uaf/util/pkipublickey.cpp |  6 +++---
 src/uaf/util/pkipublickey.h   | 15 ++++++++++-----
 2 files changed, 13 insertions(+), 8 deletions(-)

diff --git a/src/uaf/util/pkipublickey.cpp b/src/uaf/util/pkipublickey.cpp
index 37d2b95..ab43ef1 100644
--- a/src/uaf/util/pkipublickey.cpp
+++ b/src/uaf/util/pkipublickey.cpp
@@ -72,11 +72,11 @@ namespace uaf
 
     // Get the key type
     // =============================================================================================
-    PkiPublicKey::KeyType PkiPublicKey::keyType() const
+    PkiPublicKey::UaPkiKeyType PkiPublicKey::keyType() const
     {
-        if (uaPkiPublicKey_.keyType() == UaPkiPublicKey::DSA)
+        if (uaPkiPublicKey_.keyType() == UaPkiKeyType_DSA)
             return DSA;
-        else if (uaPkiPublicKey_.keyType() == UaPkiPublicKey::RSA)
+        else if (uaPkiPublicKey_.keyType() == UaPkiKeyType_RSA)
             return RSA;
         else
             return Unknown;
diff --git a/src/uaf/util/pkipublickey.h b/src/uaf/util/pkipublickey.h
index 494926b..4ef9b15 100644
--- a/src/uaf/util/pkipublickey.h
+++ b/src/uaf/util/pkipublickey.h
@@ -28,6 +28,9 @@
 #include <vector>
 #include <stdint.h>
 #include <iostream>
+
+#include <openssl/x509.h>
+
 // SDK
 #include "uapki/uapkipublickey.h"
 #include "uapki/uapkirsakeypair.h"
@@ -45,6 +48,8 @@ namespace uaf
     *
     * @ingroup Util
     ***********************************************************************************************/
+
+
     class UAF_EXPORT PkiPublicKey
     {
     public:
@@ -54,11 +59,11 @@ namespace uaf
          *
          * @ingroup Util
          */
-        enum KeyType
+        enum UaPkiKeyType
         {
-            RSA     = 0,
-            DSA     = 1,
-            Unknown = 2
+            RSA        = EVP_PKEY_RSA,     /*!< KeyType: RSA       */
+            DSA        = EVP_PKEY_DSA,     /*!< KeyType: DSA       */
+            Unknown    = EVP_PKEY_NONE     /*!< KeyType: Unknown   */
         };
 
 
@@ -101,7 +106,7 @@ namespace uaf
 
 
         /** Get the key type. */
-        KeyType keyType() const;
+        UaPkiKeyType keyType() const;
 
 
         /** Get the size of the key */
-- 
2.1.4

WimPessemier, 2015-11-17:
Hi, thanks!
Indeed these were the modifications needed, some other people have e-mailed me very similar patches but I didn't find the time yet to commit them.
In order to maintain backwards compatibility I'll add a flag to specify the SDK version (or perhaps I can guess the version based on the INCLUDE folder).
I'll apply the changes today or tomorrow.
Thanks,
Wim

WimPessemier, 2015-11-19:
Okay all tests are passing so I've merged the changes with the main master branch.

closed 2015-11-19 by WimPessemier