uaf/uaf #94: ExtensionObject is not accessible

jorrita, 2015-12-28:
Hi Wim,
in issue 53 you added support for ExtensionObject and ModelChangeStructureDataType. Now we have a service that exposes data in its address space as a struct through an ExtensionObject. When reading this data in pyuaf, we get this ExtensionObject, but there seems to be no way to access the data within:
```
>>> rd = self.meynClient.read(my_address)
>>> rd.targets[0]
<pyuaf.client.results.ReadResultTarget( - clientConnectionId : 0
                                        - status             : Good
                                        - opcUaStatusCode    : 0
                                        - data               :  - encodingTypeId : Ns=2='http://meyn.com/mlogic'|Numeric|5111
                                        - dataTypeId     : Ns=0='http://opcfoundation.org/ua'|Numeric|0
                                        - sourceTimestamp    : 2015-12-28T13:53:05.869Z
                                        - serverTimestamp    : 2015-12-28T14:36:05.126Z
                                        - sourcePicoseconds  : 0
                                        - serverPicoseconds  : 0)>
>>> dt=rd.targets[0].data
>>> dt
<pyuaf.util.ExtensionObject( - encodingTypeId : Ns=2='http://meyn.com/mlogic'|Numeric|5111
                             - dataTypeId     : Ns=0='http://opcfoundation.org/ua'|Numeric|0)>
>>> dir(dt)
['__class__', '__del__', '__delattr__', '__dict__', '__doc__', '__format__', '__getattr__', '__getattribute__', '__hash__', '__init__', '__module__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__swig_destroy__', '__swig_getmethods__', '__swig_setmethods__', '__weakref__', 'dataTypeId', 'encodingTypeId', 'this']

WimPessemier, 2015-12-28:
Hi,
indeed the implementation for generic structures is not implemented yet.

I'm not familiar with decoding this kind of data, but from the SDK docs it seems that some classes would need to be wrapped by PyUAF:
 - [UaGenericStructureValue](http://documentation.unified-automation.com/uasdkcpp/1.5.0/html/classUaGenericStructureValue.html)
 - [UaStructureDefinition](http://documentation.unified-automation.com/uasdkcpp/1.5.0/html/classUaStructureDefinition.html)
 - [UaStructureField](http://documentation.unified-automation.com/uasdkcpp/1.5.0/html/classUaStructureField.html)
 - (and perhaps a few more)

This would imply that the **structure definition** is exposed by your server (or is already known by your client). 

Is this the case? Or do you just need to access the raw data, which you will decode yourself in another way?

jorrita, 2016-01-04:
Hi,
The structure definition is available, judging from the fact that UaExpert knows what the ExtensionObject contains (in this case three Guids).
I must admit that I have no idea how much work it is to add the extra wrapping code - nor how many other users there are that use ExtensionObjects (I suspect zero). Is this something you can look into?

WimPessemier, 2016-01-04:
If the work would consist of wrapping the classes I've mentioned before, then it's not too much work and I can fit it in somewhere in the coming days/weeks. 
But to develop and test it, I need to have an example server that exposes this kind of structures. Could you verify if your structures match those of the uaservercpp? E.g. do they appear similar to the
```Demo / 000_Static / Scalar / Structures / Structure ```
node in the uaservercpp address space?

jorrita, 2016-01-04:
Yes, apart from the data type contained within (and the fact that in our case they are read-only), they look the same.

WimPessemier, 2016-01-04:
Okay I'll look into it (could be interesting for our applications too).
I'll keep you up to date.

WimPessemier, 2016-01-29:
I've implemented some classes already, but it needs some more work. Maybe next week I'll have some more time to implement the new features...

WimPessemier, 2016-02-17:
OK here's a little preview:

- [how_to_read_and_write_structures.py](https://github.com/WimPessemier/uaf/blob/master/examples/pyuaf/client/how_to_read_and_write_structures.py)

There are many things that have not been implemented so far (StructureArrays, Unions, UnionArrays, OptionSets, OptionSetArrays) but I will work on it...
Cheers,
Wim

WimPessemier, 2016-02-23:
Also structures that contain other structures (or arrays of structures) are now supported (see updated example).
The new features are only pushed to my personal clone of the master branch, as they're not documented yet. But the examples should be sufficiently instructive to understand the API already, if it's urgent...

sizun, 2016-02-26:
Hello Wim,
I am also interested in reading and writing ExtensionObject's; I used your clone to try your how_to_read_and_write_structures.py example but it fails with:
ImportError: No module named structurefielddatatypes
Best regards,
Patrick

WimPessemier, 2016-02-28:
Hi Patrick,
Hmm maybe I forgot to add some new files to the repository. Anyway there is a slightly new version in my personal fork, which should contain all files. Please pull the latest version, remove everything from the build/ folder (the lib/ folder too if you want to be sure that no old files keep hanging around), and do the full cmake installation procedure again.
Let me know if you still get the same problem. As said, the documentation has not been updated yet, but I'll try to find some time for that soon.
HTH,
Wim

sizun, 2016-02-29:
After pulling the last version, I managed to run the example on uaservercpp. Now I will try with our own server.
Thank you,
Patrick

WimPessemier, 2016-03-14:
Structures, Unions and arrays of Structures and Unions are now supported in the official master branch...

closed 2016-03-14 by WimPessemier