uaf/uaf #95: Empty array as method input arguments

daveneijsen, 2016-01-07:
Hi Wim,

I have a problem calling a method expecting an array of nodes with a certain type. When passing an EMPTY array as method input arguments, I get the following exception:

```
  File "/usr/lib/python2.7/site-packages/pyuaf/client/__init__.py", line 2851, in call
    inputArgsVector.append(item)
  File "/usr/lib/python2.7/site-packages/pyuaf/util/__init__.py", line 4439, in append
    return ___init__.VariantVector_append(self, x)
TypeError: Could not determine the array type, empty array
```

Is there a special way to instantiate an empty list in PyUAF that can be used as method argument?

Regards,
Dave

WimPessemier, 2016-01-08:
Hello Dave,
Indeed, this seems to be a bug. I can reproduce the exception so I'll put it on the to-do list...
Cheers,
Wim

WimPessemier, 2016-01-13:
Just a quick follow-up: I'm still working on a solution, but it requires some more work. I've got a working solution on my laptop already (for just 1 datatype), but I need to "copy" the solution for the other datatypes and write the unit tests etc. before committing. I'll let you know when it's ready.

WimPessemier, 2016-01-19:
There's now a working version in my personal fork (https://github.com/uaf/uaf/compare/master...WimPessemier:master).

You can now do:
```python
import pyuaf
from pyuaf.util.primitives import BooleanArray, UInt32Array, StringArray
from pyuaf.util import LocalizedTextVector

# create a client instance
c = pyuaf.client.Client()

# enable logging
settings = c.clientSettings()                                                       
settings.logToStdOutLevel = pyuaf.util.loglevels.Debug                          
c.setClientSettings(settings)

# call a method with empty arrays as input args                                                   
c.call(pyuaf.util.Address(), pyuaf.util.Address(), [BooleanArray(), UInt32Array(), StringArray(), LocalizedTextVector()] )
```

... which correctly says that the empty arrays are used:
```
2016-01-19 09:24:44.081 Debug                     Client              Processing the following MethodCallRequest:
2016-01-19 09:24:44.081 Debug                     Client               - requestHandle                    : REQUESTHANDLE_NOT_ASSIGNED
2016-01-19 09:24:44.081 Debug                     Client               - targets[]
2016-01-19 09:24:44.081 Debug                     Client                  - targets[0]
2016-01-19 09:24:44.081 Debug                     Client                     - objectAddress
2016-01-19 09:24:44.081 Debug                     Client                        - type                    : ExpandedNodeId
2016-01-19 09:24:44.081 Debug                     Client                        - expandedNodeId          : Svr=???|Ns=???|Numeric|0
2016-01-19 09:24:44.081 Debug                     Client                     - methodAddress
2016-01-19 09:24:44.081 Debug                     Client                        - type                    : ExpandedNodeId
2016-01-19 09:24:44.081 Debug                     Client                        - expandedNodeId          : Svr=???|Ns=???|Numeric|0
2016-01-19 09:24:44.081 Debug                     Client                     - inputArguments[]           : [(BooleanArray){}, (UInt32Array){}, "(StringArray){}", "(LocalizedTextArray)This array type cannot be displayed"]
...
```

It's not tested on all platforms yet, and the documentation and unittests have not been added. I'll do this when I addressed the other recent issues.

jorrita, 2016-01-19:
Tested this locally with one of our services and it works as advertised.

WimPessemier, 2016-01-29:
The issue is fixed in the latest version (2.0.3) of the official master branch. 

I could not generate the online pyuaf documentation yet, because the version of Sphinx on my Fedora system has some issues, so I must install the latest version of Sphinx first. I will do this next week.

closed 2016-01-29 by WimPessemier