uaf/uaf #96: setMonitoringMode returns "Unknown client hande ()" status and .isGood() returns false, but changing monitoring mode works

monstersintokyo, 2016-01-14:
When checking the status that *setMonitoringMode* returns, it will return *Unknown client handle (handle_no)*. Moreover checking *isGood()* returns *false*. Changing the mode still works successfully.

Tried it with UA Testserver 1.4.3 and pyuaf@2.0.2.

Example:

```python
# coding=utf-8

from time import sleep

from pyuaf.client import Client
from pyuaf.client.settings import ClientSettings
from pyuaf.util import Address, monitoringmodes, RelativePathElement, QualifiedName, ExpandedNodeId
from pyuaf.util.errors import UafError

def value_changed(notification):
    print("Value has changed: %d" % notification.data.value)

def other_value_changed(notification):
    print("Other value has changed: %d" % notification.data.value)

def test_monitoring():

    try:
        result = my_client.createMonitoredData(
            addresses = [some_node, other_node],
            notificationCallbacks = [value_changed, other_value_changed],
        )
        handles = [target.clientHandle for target in result.targets]
    except UafError:
        print("Oops, an unexpected error!")

    sleep(2)

    _set_monitoring_modes(handles)

    sleep(4)

def _set_monitoring_modes(handles):
    try:
        statuses = my_client.setMonitoringMode(handles, monitoringmodes.Sampling)
        for s in statuses:
            print s  # Will return unknown client handle
            print s.isGood()  # Will be false
    except UafError as u:
        print('ERROR: Could not set disabled monitoring mode for %s' % u)

demoNsUri = "http://www.unifiedautomation.com/DemoServer"
demoServerUri = "urn:UnifiedAutomation:UaServerCpp"

my_client = Client(ClientSettings("my_client", ["opc.tcp://localhost:48010"]))
base_node = Address(ExpandedNodeId("Demo.Dynamic", demoNsUri, demoServerUri))
some_node = Address(base_node, [RelativePathElement(QualifiedName("VariableWithNumericNodeId", demoNsUri))])
other_node = Address(base_node, [RelativePathElement(QualifiedName("VariableWithGuidNodeId", demoNsUri))])

test_monitoring()
```

WimPessemier, 2016-01-14:
Thanks for providing a full example!
I'll check it out ASAP.

WimPessemier, 2016-01-19:
Just had a look at it: it turned out to be a silly bug. Here's a patch:

```diff
diff --git a/src/uaf/client/subscriptions/subscription.cpp b/src/uaf/client/subscriptions/subscription.cpp
index e4f4443..ce51321 100644
--- a/src/uaf/client/subscriptions/subscription.cpp
+++ b/src/uaf/client/subscriptions/subscription.cpp
@@ -274,9 +274,9 @@ namespace uaf
                 uaStatusCodes);
 
         if (sdkStatus.isGood())
-            return statuscodes::Good;
+            ret = statuscodes::Good;
         else
-            return SetMonitoringModeInvocationError(sdkStatus);
+            ret = SetMonitoringModeInvocationError(sdkStatus);
 
 
         logger_->debug("Result of OPC UA service call: %s", ret.toString().c_str());
```

Instead of updating the 'ret' variable, the function returned immediately, before updating the statuses (as you noticed).

I've pushed the changes to my personal branch for now. To save some time, I will address some other issues first, before fully testing and creating a pull request for the uaf/uaf master branch.

monstersintokyo, 2016-01-20:
Thank you for taking a look at it. Glad it was just a minor issue!

WimPessemier, 2016-01-29:
The issue is fixed in the latest version (2.0.3) of the official master branch. 

I could not generate the online pyuaf documentation yet, because the version of Sphinx on my Fedora system has some issues, so I must install the latest version of Sphinx first. I will do this next week.

closed 2016-01-29 by WimPessemier