uaf/uaf #97: setMonitoringMode throws SetMonitoringModeInvocationError, when using different publishingIntervals in subscriptions

monstersintokyo, 2016-01-14:
UA Demo Server 1.4.3, pyuaf@2.0.2

First call will succeed, any following will throw the error.

```python
# coding=utf-8

from time import sleep

from pyuaf.client import Client
from pyuaf.client.settings import ClientSettings, SubscriptionSettings
from pyuaf.util import Address, monitoringmodes, RelativePathElement, QualifiedName, ExpandedNodeId
from pyuaf.util.errors import UafError

def some_value_changed(notification):
    print("Some value has changed: %d" % notification.data.value)

def other_value_changed(notification):
    print("Other value has changed: %d" % notification.data.value)

def test_monitoring_some():

    subscription_settings = SubscriptionSettings()
    subscription_settings.publishingIntervalSec = 0.2

    try:
        result = my_client.createMonitoredData(
            addresses = [some_node],
            notificationCallbacks = [some_value_changed],
            subscriptionSettings=subscription_settings
        )
        handles = [target.clientHandle for target in result.targets]
    except UafError:
        print("Oops, an unexpected error!")

    sleep(1)

    _set_monitoring_modes(handles)

    sleep(2)

def test_monitoring_other():

    subscription_settings = SubscriptionSettings()
    subscription_settings.publishingIntervalSec = 0.1

    try:
        result = my_client.createMonitoredData(
            addresses = [other_node],
            notificationCallbacks = [other_value_changed],
            subscriptionSettings=subscription_settings
        )
        handles = [target.clientHandle for target in result.targets]
    except UafError:
        print("Oops, an unexpected error!")

    sleep(1)
    _set_monitoring_modes(handles)

def _set_monitoring_modes(handles):
    try:
        statuses = my_client.setMonitoringMode(handles, monitoringmodes.Disabled)
    except UafError as u:
        print("Error while setting monitoring mode for handles {}: {} {}".format(handles, u, u.message))

demoNsUri = "http://www.unifiedautomation.com/DemoServer"
demoServerUri = "urn:UnifiedAutomation:UaServerCpp"

my_client = Client(ClientSettings("my_client", ["opc.tcp://localhost:48010"]))
base_node = Address(ExpandedNodeId("Demo.Dynamic", demoNsUri, demoServerUri))
some_node = Address(base_node, [RelativePathElement(QualifiedName("VariableWithNumericNodeId", demoNsUri))])
other_node = Address(base_node, [RelativePathElement(QualifiedName("VariableWithGuidNodeId", demoNsUri))])

test_monitoring_some()
test_monitoring_other()

```

monstersintokyo, 2016-01-14:
After further testing, changing the mode still seems to work, besides throwing the error.

WimPessemier, 2016-01-19:
Your guess was right: when the first subscription is checked and there's no matching handles, the setMonitoringMode function exits without checking the next subscription. 

The fix is very simple (basically: ``if (realSize > 0) { invoke-the-service } else { skip-this-subscription }``)  but the patch is big because the indentation of a whole code block changed, and because I've added some more logging...

I've pushed the commits to my personal forked master branch for now, I will update the uaf/uaf master branch when fully tested.

```diff
diff --git a/src/uaf/client/subscriptions/subscription.cpp b/src/uaf/client/subscriptions/subscription.cpp
index ce51321..e648707 100644
--- a/src/uaf/client/subscriptions/subscription.cpp
+++ b/src/uaf/client/subscriptions/subscription.cpp
@@ -244,6 +244,7 @@ namespace uaf
         ranks.resize(maxSize);
         ids.resize(maxSize);
         MonitoredItemsMap::const_iterator it;
+        vector<ClientHandle> handlesToChange;
         for (uint32_t i = 0; i < maxSize; i++)
         {
             it = monitoredItemsMap_.find(clientHandles[i]);
@@ -254,44 +255,58 @@ namespace uaf
 
                 ranks[realSize-1] = i;
                 ids[realSize-1]   = it->second.monitoredItemId;
+
+                handlesToChange.push_back(clientHandles[i]);
             }
         }
 
-        // don't forget to resize the ranks and ids now to their real size:
-        ranks.resize(realSize);
-        ids.resize(realSize);
-
-        // now invoke the service
-        OpcUa_MonitoringMode opcUaMonitoringMode = fromUafToSdk(monitoringMode);
-        UaClientSdk::ServiceSettings uaServiceSettings;
-        serviceSettings.toSdk(uaServiceSettings);
-        UaStatusCodeArray uaStatusCodes;
-
-        SdkStatus sdkStatus = uaSubscription_->setMonitoringMode(
-                uaServiceSettings,
-                opcUaMonitoringMode,
-                ids,
-                uaStatusCodes);
-
-        if (sdkStatus.isGood())
-            ret = statuscodes::Good;
-        else
-            ret = SetMonitoringModeInvocationError(sdkStatus);
+        if (realSize > 0)
+        {
+            logger_->debug("The following client handles were found: [%s]",
+                           uaf::uint32ArrayToString(handlesToChange).c_str());
+
+            // don't forget to resize the ranks and ids now to their real size:
+            ranks.resize(realSize);
+            ids.resize(realSize);
+
+            // now invoke the service
+            OpcUa_MonitoringMode opcUaMonitoringMode = fromUafToSdk(monitoringMode);
+            UaClientSdk::ServiceSettings uaServiceSettings;
+            serviceSettings.toSdk(uaServiceSettings);
+            UaStatusCodeArray uaStatusCodes;
+
+            SdkStatus sdkStatus = uaSubscription_->setMonitoringMode(
+                    uaServiceSettings,
+                    opcUaMonitoringMode,
+                    ids,
+                    uaStatusCodes);
+
+            if (sdkStatus.isGood())
+                ret = statuscodes::Good;
+            else
+                ret = SetMonitoringModeInvocationError(sdkStatus);
 
 
-        logger_->debug("Result of OPC UA service call: %s", ret.toString().c_str());
+            logger_->debug("Result of OPC UA service call: %s", ret.toString().c_str());
 
-        if (ret.isGood())
-        {
-            for (uint32_t i = 0; i < realSize; i++)
+            if (ret.isGood())
             {
-                if (OpcUa_IsGood(uaStatusCodes[i]))
-                    results[ranks[i]] = statuscodes::Good;
-                else
-                    results[ranks[i]] = ServerCouldNotSetMonitoringModeError(
-                            clientHandles[ranks[i]],
-                            SdkStatus(uaStatusCodes[i]));
+                for (uint32_t i = 0; i < realSize; i++)
+                {
+                    if (OpcUa_IsGood(uaStatusCodes[i]))
+                        results[ranks[i]] = statuscodes::Good;
+                    else
+                        results[ranks[i]] = ServerCouldNotSetMonitoringModeError(
+                                clientHandles[ranks[i]],
+                                SdkStatus(uaStatusCodes[i]));
+                }
             }
+
+        }
+        else
+        {
+            logger_->debug("The client handles do not belong to this subscription, skipping");
+            // ret remains Uncertain
         }
 
         return ret;
```

monstersintokyo, 2016-01-20:
Great, thank you! And I am sorry, that I could not provide a patch by myself. Still didn't have the chance to acquire any C++ skills. It would have been a mess. Hence thanks again for fixing the issue (and fixing it that fast, as always :-))

Looking forward to test it on our webserver.

WimPessemier, 2016-01-20:
No worries, I hope everything works as expected now!

monstersintokyo, 2016-01-26:
Any ideas when those changes land in an official version?

WimPessemier, 2016-01-26:
Hmmm I'm still busy with implementing https://github.com/uaf/uaf/issues/94 and https://github.com/uaf/uaf/issues/98, but this will likely take 2 or 3 more weeks. 
I'll see if I can make an "intermediary" official version in one of the next days, it's only a matter of adding some documentation and running the tests on my non-linux platforms, so it's not a lot of work. 
I'll let you know...

monstersintokyo, 2016-01-27:
Ok that'd be great!

WimPessemier, 2016-01-29:
The issue is fixed in the latest version (2.0.3) of the official master branch. 

I could not generate the online pyuaf documentation yet, because the version of Sphinx on my Fedora system has some issues, so I must install the latest version of Sphinx first. I will do this next week.

closed 2016-01-29 by WimPessemier

monstersintokyo, 2016-02-02:
Our build server is green; tested it locally and deployed it to a machine. Everything is running smoothly and as expected now.
Thanks Wim!