uaf/uaf #98: Access to ServerState

jorrita, 2016-01-19:
Is there a way to access the ServerState from the client? SessionState seems to be available, but what we want to know is not the session state, but that the server is ready to handle requests.
The SDK provides an interface for that (OpcUa_ServerState UaSession::serverState), but I found no way to access the session from pyuaf.

WimPessemier, 2016-01-19:
Okay, I didn't know about this variable. 
It will be easy to add, I'll take a look at it after issue #96 is addressed.

jorrita, 2016-01-19:
Great, thanks!
Het is ook nieuw voor mij, maar het klinkt redelijk – verbindingsstatus en serverstatus zijn verschillende dingen…
Hier wordt momenteel hard gewerkt aan ondersteuning van serverState aan de service-kant, dat schijnt ook niet triviaal te zijn.
Jorrit

WimPessemier, 2016-01-29:
The server states are implemented now (version 2.0.3).

You can request the server state using: ``myClient.sessionInformation(clientConnectionId).serverState``

See 
 - ``pyuaf.util.serverstates`` (module defining all states)
 - ``pyuaf.client.SessionInformation`` (this class has an attribute called ``serverState`` now.)

I could not generate the online pyuaf documentation yet, because the version of Sphinx on my Fedora system has some issues, so I must install the latest version of Sphinx first. I will do this next week.

closed 2016-01-29 by WimPessemier

WimPessemier, 2016-02-02:
The updated docs are online now, see
- [pyuaf.util.serverstates](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_util_serverstates.html)
- [pyuaf.util.SessionInformation](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#pyuaf.client.SessionInformation.serverState)