uaf/uaf #104: Consider building with multiple processes

robi-wan, 2016-03-21:
The article [CMake: Building with all your cores](https://blog.kitware.com/cmake-building-with-all-your-cores/) explains some approaches to accelerate the compile step. ([Some more background](https://cmake.org/pipermail/cmake/2009-April/028669.html))

For Windows and Visual Studio this means setting the compiler switch [/MP (Build with Multiple Processes)](https://msdn.microsoft.com/de-de/library/bb385193(v=vs.100).aspx).
The needed change:
```
 src/UafMacros.cmake | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/UafMacros.cmake b/src/UafMacros.cmake
index ecba964..ac56f36 100644
--- a/src/UafMacros.cmake
+++ b/src/UafMacros.cmake
@@ -76,7 +76,7 @@ ENDMACRO(setUnifiedAutomationSdkCompilerDir)
 MACRO(setUafCompilerFlags)
 
     if (WIN32)
-        set(CMAKE_CXX_FLAGS "/EHsc")
+      set(CMAKE_CXX_FLAGS "/EHsc /MP")
         add_definitions(-D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -DUNICODE -D_UNICODE -D_UA_STACK_USE_DLL)  
     else (WIN32)
         if (FORCE32)
```

The /MP switch takes an optional argument: processMax
> The maximum number of processes that the compiler can create. [..] If you omit the processMax argument, the compiler retrieves the number of effective processors on your computer from the operating system, and creates a process for each processor.

So omitting the number is fine.

For Linux the corresponding setting is the `-j` switch: [The documentation](https://www.gnu.org/software/make/manual/make.html#Parallel) says it is also possible to omit the integer argument to that switch - but I cannot say how that impacts the build (system).

WimPessemier, 2016-03-22:
Thanks a lot for the tip/background/patch! I'll include it in the next minor version (perhaps multi-core by default, and an optional CMake flag to make it single-core... will test first.)

WimPessemier, 2016-04-24:
I've added a CMake flag (see https://github.com/uaf/uaf/pull/108/commits/52910fbafaafe212821d365a763c8b3befb005f5)
from the changelog:  

> On MS Windows, multi-core compiling is now possible on via a CMake flag (BUILD_WITH_MULTIPLE_PROCESSES). 
By default, this flag is OFF, so there is no change in behavior. By adding -DBUILD_WITH_MULTIPLE_PROCESSES=ON to the command line, the build process will be faster if a multi-core CPU is available. 
On Linux, multi-core compilation is not specified with the cmake command, but with the make command ("make -j"), so the BUILD_WITH_MULTIPLE_PROCESSES flag is not allowed for this platform.

On linux it works great (make -j), the build is much faster.

On Windows I can see that the /MP flag is accepted by Visual Studio, but the build is still sequential (1 process). I think it's because I'm running Windows inside a VM (despite assigning 4 of the 8 available CPUs in my Virtualbox). If you see it doesn't work for you, let me know...

closed 2016-04-24 by WimPessemier

robi-wan, 2016-04-26:
I run our CI builds with the new CMake flag - it accelerates the compilation.
It works!