uaf/uaf #105: subscriptions can't be removed

jorrita, 2016-03-22:
There is an interface to manuallyUnsubscribe a subscription, but this does not do what I expected -- the deleted subscription seems to promptly reappear. 
I used easiest_client_tutorial.py to show this, with the following lines instead of "del myClient":
```
print myClient.allSubscriptionInformations()
myClient.manuallyUnsubscribe(0,0)    
time.sleep(1)
print myClient.allSubscriptionInformations()
time.sleep(30)
print myClient.allSubscriptionInformations()
```
Surprisingly, we seem to get a new subscription:
```
Synchronous creating monitored data items
=========================================

Iteration  0
Waiting for notifications...
 --> received new Byte value: 247
 --> received new Float value: 85381
 --> received new Byte value: 3
 --> received new Float value: 85394
 --> received new Byte value: 15
 --> received new Float value: 85407
 --> received new Byte value: 27
 --> received new Float value: 85420
[<pyuaf.client.SubscriptionInformation( - clientConnectionId : 0
                                       - clientSubscriptionHandle: 0
                                       - subscriptionState  : 0 (Created)
                                       - subscriptionSettings
                                          - unique          : false
                                          - publishingIntervalSec: 1
                                          - lifeTimeCount   : 1200
                                          - maxKeepAliveCount: 5
                                          - maxNotificationsPerPublish: 0
                                          - priority        : 0)>]
 --> received new Byte value: 40
 --> received new Float value: 85434
[]
 --> received new Byte value: 239
 --> received new Float value: 85654
 --> received new Byte value: 251
  --> received new Float value: 85667
 --> received new Byte value: 7
 --> received new Float value: 85680
 --> received new Byte value: 20
 --> received new Float value: 85695
 --> received new Byte value: 32
 --> received new Float value: 85708
 --> received new Byte value: 44
 --> received new Float value: 85721
 --> received new Byte value: 57 --> received new Float value: 85736

 --> received new Byte value: 69
 --> received new Float value: 85749
 --> received new Byte value: 81
 --> received new Float value: 85762
 --> received new Byte value: 93
 --> received new Float value: 85775
 --> received new Byte value: 106
 --> received new Float value: 85789
 --> received new Byte value: 118
 --> received new Float value: 85803
 --> received new Byte value: 130
 --> received new Float value: 85816
 --> received new Byte value: 142
  --> received new Float value: 85829
 --> received new Byte value: 155
 --> received new Float value: 85843
[<pyuaf.client.SubscriptionInformation( - clientConnectionId : 0
                                       - clientSubscriptionHandle: 1
                                       - subscriptionState  : 0 (Created)
                                       - subscriptionSettings
                                          - unique          : false
                                          - publishingIntervalSec: 1
                                          - lifeTimeCount   : 1200
                                          - maxKeepAliveCount: 5
                                          - maxNotificationsPerPublish: 0
                                          - priority        : 0)>]
>>> 
```

WimPessemier, 2016-03-22:
Indeed, this is a bug. In the UAF, the requests to create monitored items are "persistent". It means that if a server goes down, or if you start to monitor items on a server that is not online yet, the request is stored in the client memory, so it can be re-created automatically when the server comes online (again). 
Due to the bug, the ``manuallyUnsubscribe`` method only deleted the subscription, but not the associated persistent request in memory. The subscription was therefore re-created automatically.

Below is a preliminary patch. 

```patch
diff --git a/src/uaf/client/subscriptions/subscription.cpp b/src/uaf/client/subscriptions/subscription.cpp
index e648707..79e29df 100644
--- a/src/uaf/client/subscriptions/subscription.cpp
+++ b/src/uaf/client/subscriptions/subscription.cpp
@@ -124,7 +124,7 @@ namespace uaf
     }
 
 
-    Status Subscription::deleteSubscription()
+    Status Subscription::deleteSubscription(bool deletePersistentRequest)
     {
         Status ret;
 
@@ -162,17 +162,25 @@ namespace uaf
             // remove the notification buffer according to the kind
             if (it->second.settings.kind() == MonitoredItemSettings::Data)
             {
-                database_->createMonitoredDataRequestStore.updateTargetStatus(
-                        it->second.requestHandle,
-                        it->second.targetRank,
-                        SubscriptionHasBeenDeletedError());
+                if (deletePersistentRequest)
+                    database_->createMonitoredDataRequestStore.remove(
+                            it->second.requestHandle);
+                else
+                    database_->createMonitoredDataRequestStore.updateTargetStatus(
+                            it->second.requestHandle,
+                            it->second.targetRank,
+                            SubscriptionHasBeenDeletedError());
             }
             else
             {
-                database_->createMonitoredEventsRequestStore.updateTargetStatus(
-                        it->second.requestHandle,
-                        it->second.targetRank,
-                        SubscriptionHasBeenDeletedError());
+                if (deletePersistentRequest)
+                    database_->createMonitoredEventsRequestStore.remove(
+                            it->second.requestHandle);
+                else
+                    database_->createMonitoredEventsRequestStore.updateTargetStatus(
+                            it->second.requestHandle,
+                            it->second.targetRank,
+                            SubscriptionHasBeenDeletedError());
             }
 
             // remove the monitoredItemsMap_ entry
diff --git a/src/uaf/client/subscriptions/subscription.h b/src/uaf/client/subscriptions/subscription.h
index 9ce46f1..e9c51ca 100644
--- a/src/uaf/client/subscriptions/subscription.h
+++ b/src/uaf/client/subscriptions/subscription.h
@@ -104,9 +104,11 @@ namespace uaf
         /**
          * Delete the subscription between client and server.
          *
+         * @param deleteRequest  True to delete the persistent request that
+         *                       created the subscription.
          * @return  Good if the subscription was deleted.
          */
-        uaf::Status deleteSubscription();
+        uaf::Status deleteSubscription(bool deletePersistentRequest = false);
 
 
         /**
diff --git a/src/uaf/client/subscriptions/subscriptionfactory.cpp b/src/uaf/client/subscriptions/subscriptionfactory.cpp
index 0ccbd2e..d6b6c5c 100644
--- a/src/uaf/client/subscriptions/subscriptionfactory.cpp
+++ b/src/uaf/client/subscriptions/subscriptionfactory.cpp
@@ -134,7 +134,7 @@ namespace uaf
 
         if (acquisitionStatus.isGood())
         {
-            ret = subscription->deleteSubscription();
+            ret = subscription->deleteSubscription(true); // the unsubscribing is planned, so we can remove the persistent request
             releaseSubscription(subscription);
         }
         else
```

WimPessemier, 2016-04-24:
The bugfix is now included in the master branch,

closed 2016-04-24 by WimPessemier