uaf/uaf #106: memory management problem

jorrita, 2016-03-22:
After adapting easiest_client_tutorial.py to remove all steps except monitoredData and adding a loop  as shown below, I reliably get a memory problem:
```
for i in range(100):
  print "Iteration ", i
  result = myClient.createMonitoredData( addresses             = [address_Byte, address_Float],
                                         notificationCallbacks = [byteCallback, floatCallback] )

  # let's wait a second, so that a number of notifications can arrive:
  print("Waiting for notifications...")
  time.sleep(5.0)

  del myClient 
  time.sleep(1) 
  myClient = Client(myClientSettings) 
```

Output on my system:
```
[..]
Iteration  51
Waiting for notifications...
 --> received new Byte value: 187
 --> received new Float value: 50629
 --> received new Byte value: 199
 --> received new Float value: 50642
 --> received new Byte value: 212
 --> received new Float value: 50656
 --> received new Byte value: 224
 --> received new Float value: 50669
*** Error in `python': double free or corruption (!prev): 0x00000000027c0970 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x77d9e)[0x7fd842328d9e]
/lib64/libc.so.6(cfree+0x5b5)[0x7fd8423349f5]
/lib64/libpython2.7.so.1.0(+0x7ffc9)[0x7fd843018fc9]
/lib64/libpython2.7.so.1.0(_PyObject_GenericGetAttrWithDict+0x228)[0x7fd84301fe88]
/lib64/libpython2.7.so.1.0(+0xa4198)[0x7fd84303d198]
/lib64/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x1c1d)[0x7fd84307770d]
/lib64/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x820)[0x7fd84307c410]
/lib64/libpython2.7.so.1.0(+0x6f5dc)[0x7fd8430085dc]
/lib64/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7fd842fe3903]
/lib64/libpython2.7.so.1.0(+0x59815)[0x7fd842ff2815]
/lib64/libpython2.7.so.1.0(PyObject_Call+0x43)[0x7fd842fe3903]
/lib64/libpython2.7.so.1.0(+0x4a9e5)[0x7fd842fe39e5]
/lib64/libpython2.7.so.1.0(PyObject_CallMethod+0xa9)[0x7fd842fe3ce9]
/usr/lib/python2.7/site-packages/pyuaf/client/___init__.so(_ZN23SwigDirector_ClientBase19dataChangesReceivedESt6vectorIN3uaf22DataChangeNotificationESaIS2_EE+0x2ea)[0x7fd83436ec4a]
/usr/lib/python2.7/site-packages/libuafclient.so(_ZN3uaf12Subscription10dataChangeERK28UaMonitoredItemNotificationsRK17UaDiagnosticInfos+0x6c4)[0x7fd8340a6394]
/usr/lib/python2.7/site-packages/libuafclient.so(_ZN3uaf19SubscriptionFactory10dataChangeEjRK28UaMonitoredItemNotificationsRK17UaDiagnosticInfos+0x71)[0x7fd83409f8e1]
/opt/unifiedautomation/sdk/lib64/libuaclient.so(_ZN11UaClientSdk16UaSessionPrivate15publishCallbackEP22_OpcUa_PublishResponseRKSt4listI34_OpcUa_SubscriptionAcknowledgementSaIS4_EE+0xae7)[0x7fd833cd7f97]
/opt/unifiedautomation/sdk/lib64/libuaclient.so(_ZN11UaClientSdk11CallbackJob7executeEv+0x110)[0x7fd833cd9d00]
/opt/unifiedautomation/sdk/lib64/libuabase.so(_ZN11UaJobThread3runEv+0x50)[0x7fd8399e9300]
/opt/unifiedautomation/sdk/lib64/libuastack.so(InternalThreadMain+0x18)[0x7fd83923b108]
/opt/unifiedautomation/sdk/lib64/libuastack.so(pthread_start+0x21)[0x7fd839247611]
/lib64/libpthread.so.0(+0x752a)[0x7fd842d8452a]
/lib64/libc.so.6(clone+0x6d)[0x7fd8423b122d]
======= Memory map: ========
[...]
```

Am I doing something wrong?

WimPessemier, 2016-03-28:
Apparently It has something to do with the order of destruction of the SWIG python Client object and the wrapped C++  Client object. If the python object is destructed just as a new notification comes in, the memory problems appear.
The easiest solution I found is to first delete all sessions, wait a little bit for any ongoing callbacks to start, and then destruct the client. See patch below.

With this patch, I've ran over 2000 iterations, with 10 notifications per second, and randomly varying destruction times (between 3 and 5 seconds after the client is created), and it cannot fault it. 
I will include the patch in the upcoming new version.

```patch
diff --git a/src/pyuaf/client/client.py b/src/pyuaf/client/client.py
index aaa631e..6481b38 100644
--- a/src/pyuaf/client/client.py
+++ b/src/pyuaf/client/client.py
@@ -76,6 +76,15 @@ class Client(ClientBase):
         self.__eventNotificationLock__ = threading.Lock()
     
     
+    def __del__(self):
+        # first disconnect all sessions, to avoid memory problems when the client
+        # is garbage collected before any ongoing callback threads are fired!
+        self.manuallyDisconnectAllSessions()
+        # wait some time for any ongoing callback threads to be fired
+        time.sleep(0.1)
+        ClientBase.__del__(self)
+    
+    
     def __dispatch_readComplete__(self, result):
         """
         Dispatch the result of the asynchronous request either to a virtual readComplete function,
@@ -1095,6 +1104,17 @@ class Client(ClientBase):
        ClientBase.manuallyDisconnect(self, clientConnectionId).test()
     
     
+    def manuallyDisconnectAllSessions(self):
+       """
+       Disconnect all sessions.
+      
+       To stress that normally the UAF takes care of session connection and disconnection,
+       this method has a "manually" prefix. Normally it should not be used explicitely,
+       as all sessions will be disconnected automatically when the client is deleted.
+       """
+       ClientBase.manuallyDisconnectAllSessions(self)
+    
+    
     def manuallySubscribe(self, clientConnectionId, subscriptionSettings=None):
        """
        Create a subscription manually.
diff --git a/src/pyuaf/client/client___init__.i b/src/pyuaf/client/client___init__.i
index c12bf7d..80411be 100644
--- a/src/pyuaf/client/client___init__.i
+++ b/src/pyuaf/client/client___init__.i
@@ -73,6 +73,7 @@
 // add some import stuff to the __init__.py file that will be produced
 %pythoncode %{
 import threading
+import time
 %}
 
 
diff --git a/src/uaf/client/client.cpp b/src/uaf/client/client.cpp
index c9d6d61..67a1455 100644
--- a/src/uaf/client/client.cpp
+++ b/src/uaf/client/client.cpp
@@ -645,6 +645,14 @@ namespace uaf
     }
 
 
+    // Manual disconnection of all sessions
+    //==============================================================================================
+    void Client::manuallyDisconnectAllSessions()
+    {
+        return sessionFactory_->deleteAllSessions();
+    }
+
+
     // Manual subscription
     //==============================================================================================
     Status Client::manuallySubscribe(
diff --git a/src/uaf/client/client.h b/src/uaf/client/client.h
index be78f83..388860e 100644
--- a/src/uaf/client/client.h
+++ b/src/uaf/client/client.h
@@ -846,6 +846,16 @@ namespace uaf
         uaf::Status manuallyDisconnect(uaf::ClientConnectionId clientConnectionId);
 
 
+        /**
+         * Disconnect all sessions.
+         *
+         * To stress that normally the UAF takes care of session connection and disconnection,
+         * this method has a "manually" prefix. Normally it should not be used explicitely,
+         * as all sessions will be disconnected automatically when the client is deleted.
+         */
+        void manuallyDisconnectAllSessions();
+
+
         ///@} //////////////////////////////////////////////////////////////////////////////////////
         /**
          *  @name SessionInformation
```

WimPessemier, 2016-04-24:
The bugfix is now included in the master branch,

closed 2016-04-24 by WimPessemier