uaf/uaf #107: Build fails for C++ examples

sunny1986, 2016-03-31:
Hi,

I am testing on Windows and I tried to build the C++ examples in Visual C++ but the build fails. It throws a total of 148 errors. Few as shown below

[
[image]
](url)

I am exactly going by step 7 to build and run C++ examples. Do I have to set any environment variables like you have shown for building the framework in step 6? Since I am able to build the framework successfully but not the C++ part.

Also as per your installation procedure in step 7 the uaf-test.sln is not the right file but its the uaf-examples.sln file. Am I trying to build a wrong file here and is that why the build fails? Since I could not find any file named uaf-test.sln in "C:\Users\XXX\uaf\examples\uaf\build"

Kindly let me know if I am missing something here since I would like to work with C++ examples but not the Python examples. Thanks!

sunny1986, 2016-04-05:
@robi-wan @WimPessemier - Can you suggest me some solutions to the above post?

Thanks!

WimPessemier, 2016-04-05:
Hi, sorry I'm commissioning some control systems abroad this week, so I have very little time at the moment...
The examples I've only tested in Linux, so perhaps you're the first one to try them in Windows.
I'll do my best to take a look at it asap.

WimPessemier, 2016-04-05:
The problem is that the C++ examples were never updated to version 2. Between v1 and v2 there are lot's of syntax changes, which is what you are seeing.
Unfortunately I really don't have the time now to migrate the examples to version 2, so I will do it next week (or perhaps during my flight back home).
Anyway for the easiest_client_example, here's the fix:
```diff
diff --git a/examples/uaf/src/client/easiest_client_example.cpp b/examples/uaf/src/client/easiest_client_example.cpp
index 5c6c413..3c4e823 100644
--- a/examples/uaf/src/client/easiest_client_example.cpp
+++ b/examples/uaf/src/client/easiest_client_example.cpp
@@ -35,9 +35,11 @@ int main(int argc, char* argv[])
     // read the Value attribute of the node that we just addressed:
     ReadResult result;
     Status status = myClient.read(nodesToBeRead,
-                                  attributeids::Value, // read the Value attribute
-                                  ReadConfig(),        // default settings for the Read service
-                                  SessionConfig(),     // default settings for the sessions to be created
+                                  attributeids::Value,                       // read the Value attribute
+                                  uaf::constants::CLIENTHANDLE_NOT_ASSIGNED, // let the UAF pick a suitable Session
+                                  NULL,                                      // default settings for the Read service
+                                  NULL,                                      // default settings for the Translate service
+                                  NULL,                                      // default session settings
                                   result);

     if (status.isGood())
```

WimPessemier, 2016-04-24:
The examples have been updated, just pull the latest version of the official master branch.

closed 2016-04-24 by WimPessemier