uaf/uaf #110: Monitoring: How to get callback on unchanged monitored value?

Atanamo, 2016-05-03:
Hi Wim,

I recently wondered how data monitoring works in detail:

I monitoring an item, which is a simple integer value to get appropriate change callbacks 
(Using the `createMonitoredData` method).

But the callback is only triggered, if the value really changes - e.g. from 1 to 2.
It doesn't seem to be called, if the item is only updated with the **same** value - e.g. setting it from 3 to 3.

Is this possible anyways?
May I have to use another mechanism or a special configuration - I didn't found an appropriate flag so far.

Hope you can help me - thanks!

WimPessemier, 2016-05-03:
Hi,

No worries, here's the short version:

- Callbacks are indeed only fired whenever a value changes. 
- This means that you cannot be notified when a value is overwritten by the same value. However, if you don't receive any callback, then you can conclude that the value has not changed
- To be sure that your client has not missed any DataChange-notifications, OPC UA offers a heartbeat mechanism for subscriptions. You just have to set the [maxKeepAliveCount](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client_settings.html#pyuaf.client.settings.SubscriptionSettings.maxKeepAliveCount), and then [register](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#pyuaf.client.Client.registerKeepAliveCallback) for KeepAlive callbacks. As long as you receive [DataChangeNotification](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#class-datachangenotification)s **or** [KeepAliveNotification](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#class-keepalivenotification)s, then your subscription is healthy.

For more elaborate information, check out a new example I've added just now: [how_to_monitor_data_changes.py](https://github.com/uaf/uaf/blob/master/examples/pyuaf/client/how_to_monitor_data_changes.py)

More general info about subscriptions:
- http://documentation.unified-automation.com/uasdkcpp/1.5.1/html/L2UaSubscription.html
- https://www.prosysopc.com/blog/opc-ua-sessions-subscriptions-and-timeouts/
- or the [green book](http://www.springer.com/us/book/9783540688983) or the UA specs, of course.

Hope this helps,

Wim

Atanamo, 2016-05-04:
Hi,

ah, the example is very useful - this helped a lot :-)
So I've to use the more basic methods instead of ``createMonitoredData``. Well, this works.

Thank you!

WimPessemier, 2016-05-04:
OK glad it helped!

closed 2016-05-04 by WimPessemier