uaf/uaf #112: OpenSSL libraries could not be found

Johen8, 2016-05-04:
While trying to build the program I got this error, don't really know what to do with it.
[image]

WimPessemier, 2016-05-04:
Hi, you're using an unsupported compiler.
The Unified Automation SDK evaluation version is only available for VS2010 SP1 and VS2012 SP4.

felixkretschmer, 2016-09-01:
Same here.
But I got vs2012 and the update 4 installed. The variable is still set to `vs2012 `and not `vs2012sp4`.
Where can I change the variable? (I'd prefer to change it, instead of changing the UA SDK folder name, due to other dependencies).

WimPessemier, 2016-09-01:
OK, I will update it in the next minor version. 
(Currently I'm finishing the support for matrices -- https://github.com/uaf/uaf/issues/113)

WimPessemier, 2016-09-20:
Should be fixed now in the latest version (2.2.1).

This was tested on:
- linux 64-bit gcc 5.3.1 + swig 3.0.7
- linux 64-bit gcc 5.3.1 + swig 3.0.10
- windows 32-bit vs2010 + swig 3.0.8
- windows 32-bit vs2010 + swig 3.0.10
- windows 64-bit vs2012sp4 (32-bit) + swig 3.0.10

closed 2016-09-20 by WimPessemier