uaf/uaf #113: How to read 2D or 3D arrays?

sizun, 2016-06-20:
Hello,

How can I read a 2D or 3D array (matrix) with PyUaf?
While reading the node "Demo.Static.Arrays.SByte" of uaservercpp returns a list of SByte,
 reading "Demo.Static.Matrix.SByte", which is of ValueRank 3, seems to return a single SByte.

Best regards,
Patrick

WimPessemier, 2016-06-20:
Hi, 
Matrices are currently not wrapped. If you need it I can put it on the to do list however, shouldn't be that much work.
Kind regards,
Wim

sizun, 2016-06-20:
Hello Wim,
Thanks for the quick answer. I connect to a server with some 2D matrices, so yes, if it could be implemented that would be great.
Cheers, Patrick

WimPessemier, 2016-09-20:
Matrices are now supported in the latest version (2.2.1).

I've created a convenience class pyuaf.util.Matrix: [see the docs](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_util.html#class-matrix).

And here's a simple example: [how_to_read_and_write_matrices.py](https://github.com/uaf/uaf/blob/master/examples/pyuaf/client/how_to_read_and_write_matrices.py).

This was tested on:
- linux 64-bit gcc 5.3.1 + swig 3.0.7
- linux 64-bit gcc 5.3.1 + swig 3.0.10
- windows 32-bit vs2010 + swig 3.0.8
- windows 32-bit vs2010 + swig 3.0.10
- windows 64-bit vs2012sp4 (32-bit) + swig 3.0.10

closed 2016-09-20 by WimPessemier