uaf/uaf #114: No module ___ini__ when running example

ku3bl3r, 2016-08-18:
Hi,

when running the "easiest_example" the first time I get this error code below. Been trying to fix it for a while for myself now. Maybe someone has a hint.

Win7 64bit
Python 2.7.12 64bit 
Built with vs2012sp4

Thanks - Karl

> Traceback (most recent call last):
  File "easiest_client_example.py", line 5, in <module>
    import pyuaf
  File "D:\myData\Coding\Python\PyCharm\UAF_Project\lib\pyuaf\__init__.py", line 2, in <module>
    from pyuaf import util, client
  File "D:\myData\Coding\Python\PyCharm\UAF_Project\lib\pyuaf\util\__init__.py", line 21, in <module>
    ___init__ = swig_import_helper()
  File "D:\myData\Coding\Python\PyCharm\UAF_Project\lib\pyuaf\util\__init__.py", line 20, in swig_import_helper
    return importlib.import_module('___init__')
  File "C:\Python27_x64\lib\importlib\__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named ___init__

WimPessemier, 2016-08-18:
Hi Karl,

Could you verify that your pyuaf/lib (sub)folders look similar to the screenshots below? This is to make sure that the build process succeeded.
Also, could you verify that the PYTHONPATH and PATH variables contain the pyuaf/lib folder? Just post the output of these commands in command prompt:
```
echo %PATH%
echo %PYTHONPATH%
```
Regards,
Wim
 
[image]
[image]
[image]
[image]

ku3bl3r, 2016-08-18:
Hi Wim,

thanks for the fast reply.

1. Yes both PATHS are set to "D:\myData\Coding\Python\PyCharm\UAF_Project\examples\..\lib" before I run the Py script
2. All files seem to be there: I'm missing the compiled Python files (pyc) and got .cxx files instead, see screenshot for pyuaf/client directory below

I guess both points are valid.

[image]

ku3bl3r, 2016-08-18:
Hi Wim,

I found this as a soluation, the swig .i file should be changed though.

OLD: return importlib.import_module('___init__')
NEW: return importlib.import_module('.___init__', 'pyuaf.util')

But I'm wondering why no one else stepped into this...

Cheers,
Karl

WimPessemier, 2016-08-18:
Hmm by googling around a bit, I think it may be a backwards incompatibility of SWIG. Apparently SWIG recently changed the way how libraries are imported, which may cause some troubles (http://www.swig.org/Release/CHANGES.current)
I assume you're using the latest version (3.10)? I will update my SWIG version and find a solution...

ku3bl3r, 2016-08-19:
Hi Wim,
that seems to be the point! Yes, I'm using SWIG 3.0.10.

WimPessemier, 2016-09-20:
Should be fixed now in the latest version. 

This was tested on:
- linux 64-bit gcc 5.3.1 + swig 3.0.7
- linux 64-bit gcc 5.3.1 + swig 3.0.10
- windows 32-bit vs2010 + swig 3.0.8
- windows 32-bit vs2010 + swig 3.0.10
- windows 64-bit vs2012sp4 (32-bit) + swig 3.0.10

closed 2016-09-20 by WimPessemier