uaf/uaf #115: ptr_fun in namespacearray.cpp

felixkretschmer, 2016-09-02:
Hi,
I try to implement a continous integration of uaf at our build server.
After the source files are directly pulled from github I don't want to do any modifaction at the source files in between.
However I needed to edit line 157 in namespacearray.cpp in order to fix the deprecated function "ptr_fun":
OLD:
`transform(ret.begin(), ret.end(), ret.begin(), std::ptr_fun<int, int>(tolower));`
NEW:
`transform(ret.begin(), ret.end(), ret.begin(), std::tolower);`
Someone aggrees on this? So I might commit this.
Regards,
Felix

WimPessemier, 2016-09-02:
Hi Felix,

Yes the very same fix was suggested to me earlier this week, by someone from the Stuttgart university who encountered the same problem.  I guess you also use VS2012sp4.

On Linux, the compiler doesn't like this fix however:  it complain that std::lower is overloaded (apparently there is a tolower in the std and root namespaces (due to <ctype.h>).
So on Linux
  ```transform(ret.begin(), ret.end(), ret.begin(), ::tolower);```
appears to do the trick, but I haven't tested this on Windows yet.

Regards,
Wim

felixkretschmer, 2016-09-02:
Hi Wim,
hehe.. yeah, Karl, I know. :)
I'll keep going an add the Linux build steps to our server in a couple of days. So I will get to this problem then probably.
Preprocessor definitions might be a solution. But we'll see. I'll keep you updated.
Regards,
Felix

WimPessemier, 2016-09-20:
Should be fixed now in the latest version.

This version addresses 4 issues:
- Matrices are now supported (https://github.com/uaf/uaf/issues/113)
- Visual studio 2012 SP4 now supported (https://github.com/uaf/uaf/issues/112)
- Latest SWIG (3.0.10) now supported (https://github.com/uaf/uaf/issues/114)
- Obsoiete C++ code in namespacearray.cpp now fixed (https://github.com/uaf/uaf/issues/115)

This was tested on:
- linux 64-bit gcc 5.3.1 + swig 3.0.7
- linux 64-bit gcc 5.3.1 + swig 3.0.10
- windows 32-bit vs2010 + swig 3.0.8
- windows 32-bit vs2010 + swig 3.0.10
- windows 64-bit vs2012sp4 (32-bit) + swig 3.0.10

closed 2016-09-20 by WimPessemier