uaf/uaf #118: How to override/disable OpcUa_BadCertificateHostNameInvalid?

bveldhoen, 2016-09-29:
When the dns in the altSubjectName of a server certificate (i.e. the hostname) does not correspond to the hostname in the url of the service, the ua sdk calls UaClientSdk::UaSessionCallback::connectError with status OpcUa_BadCertificateHostNameInvalid. This allows for client code to decide how to treat this error.
The current implementation of SessionFactory::connectError always returns false, and there does not seem to be a way to override/disable this OpcUa_BadCertificateHostNameInvalid error.
We need to, in certain circumstances, be able to override/disable this error. Do you have any pointer on how to make this possible?
Thanks!

WimPessemier, 2016-09-29:
OK I think the best solution is that SessionFactory::connectError calls another callback function at the end (instead of simply returning False). 

The idea is that you could do something like
```python
def connectError(clientConnectionId, serviceType, status, clientSideError):
    if (...):
       return True
    else:
       return False

c = Client()
c.registerConnectErrorCallback(connectError)
```

Would this work for you?

bveldhoen, 2016-09-29:
Yes, this would solve our problem!

WimPessemier, 2016-09-29:
Fixed, see: 
 - [Client.connectErrorReceived](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#pyuaf.client.Client.connectErrorReceived)
 - [Client.registerConnectErrorCallback](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#pyuaf.client.Client.registerConnectErrorCallback)
 - [Client.unregisterConnectErrorCallback](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#pyuaf.client.Client.unregisterConnectErrorCallback)

I have not tried to find ways to invoke a ConnectError, but I guess you will tell me if it doesn't work as expected :). The normal behavior is still the same (the unittests passed on Linux and Windows).

closed 2016-09-29 by WimPessemier

bveldhoen, 2016-09-30:
Thanks Wim, we will try it out asap!