uaf/uaf #121: TRL

RomiPomi, 2017-03-07:
Hallo WimPessemier

actually i am doing my master in 'industrial informatics' at the university of Emden. 

In this context i compare existing Opc Ua implementations to create an overview. One of the criteria is the Technology Readiness Level (TRL). 
Two of these implementations are OPC UA .NET and  OPC UA ANSIC which are developed by your foundation. To compare the TRL of all implementations i would be glad if you could give me a feedback with the TRL of each. The base for the TRL should be a publication by the European Commission (https://ec.europa.eu/research/participants/data/ref/h2020/wp/2014_2015/annexes/h2020-wp1415-annex-g-trl_en.pdf).

Thanks for your efforts.

WimPessemier, 2017-03-30:
Hi, 
sorry for the late reply but obviously the UAF is written on top of the Unified Automation SDK, hence the TRL should be assessed for the SDK and not for the UAF.
Regards,
Wim

closed 2017-03-30 by WimPessemier

DirkHoffmann, 2017-03-30:
Yes, UAF is a wrapper for OPC UA rather than a (standalone) implementation.
But, @RomiPomi, I would be interested in the outcome of your report when it is finished. Can you make it accessible publicly or contact me e.g. over [LinkedIn](https://www.linkedin.com/in/dirkhoffmann/)?