uaf/uaf #122: Problems with nested lists in method call

jorrita, 2017-06-08:
We have a service that exposes one method call to external clients. This method takes a struct with nested lists as follows:
<pre>
Flocks[]
            GrowerName
            GrowerExternalReference
            HouseExternalReference
            FlockExternalReference
            CustomFields[]   (contains 0..5 items of type String)
            Lots[]
                             DisplayName
                             ExternalReference
                             PlannedArrivalTime
                             PlannedAverageWeight
                             PlannedNumberOfProducts
                             PlannedTotalWeight
                             CustomFields[]   (contains 0..5 items of type String)
                             LineNumber
                             SequenceNumber
</pre>

Now when I try to create VariantVectors to encapsulate the above struct, I get an error:
<pre>
def mkLot():
  l = pyuaf.util.VariantVector()
  l.append(pyuaf.util.primitives.String("LotName"))
  l.append(pyuaf.util.primitives.String("LotExternalReference"))
  l.append(pyuaf.util.DateTime().now())
  l.append(pyuaf.util.primitives.Double(1000))  #avg weight
  l.append(pyuaf.util.primitives.Double(1000))  #PlannedNumberOfProducts
  l.append(pyuaf.util.primitives.Double(1000))  #PlannedTotalWeight
  c = pyuaf.util.VariantVector()
  c.append(pyuaf.util.primitives.String("lotcustomf1"))
  l.append(c)     # LotCustomFields
  l.append(pyuaf.util.primitives.UInt32(1))      #LineNumber
  l.append(pyuaf.util.primitives.UInt32(1))      #SequenceNumber 
  return l

def mkFlock():
  f = pyuaf.util.VariantVector()
  f.append(pyuaf.util.primitives.String( "Grower" ))
  f.append(pyuaf.util.primitives.String( "GrowerRef"))
  f.append(pyuaf.util.primitives.String( "HouseRef"))
  f.append(pyuaf.util.primitives.String( "FlockRef"))
  c = pyuaf.util.VariantVector()
  c.append(pyuaf.util.primitives.String("flockcustomf1"))
  f.append(c)     # FlockCustomFields
  lotVec =  pyuaf.util.VariantVector()
  lotVec.append(mkLot())  # <== this goes wrong!
  f.append(lotVec)
  return f

>>> flock1 = mkFlock()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 9, in mkFlock
  File "/usr/lib/python2.7/site-packages/pyuaf/util/__init__.py", line 2662, in append
    def append(self, *args): return _util.VariantVector_append(self, *args)
TypeError: Sequence contains inconsistent types!
</pre>

Some more playing around with VariantVectors gives me an idea about what may be happening:
<pre>
>>> v1=pyuaf.util.VariantVector()
>>> v2=pyuaf.util.VariantVector()
>>> v2.append(pyuaf.util.primitives.Double(1000))
>>> v1.append(v2)
>>> type(v2)
[class 'pyuaf.util.VariantVector']
>>> type(v1[0])   ## This yields a surprising result - I was expecting a VariantVector here!
[type 'list']
</pre>

Am I doing something wrong here?

WimPessemier, 2017-06-09:
Hi Jorrit, you're doing nothing wrong, there is indeed a bug. Thanks for the instructions to reproduce the problem, it's very clear.
I've started to implement a fix, but I need a little bit more time. I don't think it's a lot of work, but I'm off for holidays in the next 2 weeks so I don't know when it'll be finished.
But I'll make a fix, don't worry ;)

WimPessemier, 2017-07-11:
Just a notice that I'm still working on it, but it'll take some more time.
Basically the current type conversions ([SWIG timemaps](https://github.com/uaf/uaf/blob/master/src/pyuaf/typemaps/util_variant_python_typemap.h)  need to be changed from preprocessor macros to C++ functions. Because, Variant arrays can only be converted recursively (since variant vectors may contain other variant vectors), which was never the case before. Since macros cannot be recursive, I need to change them to functions, so it takes some time. But the result will be nicer, as the generated SWIG code will be smaller.

Anyway the typemaps were perhaps a bad design decision in PyUAF, since they are complicated and they add little value.  If somewhere in the future I will need to make breaking API changes due to SDK API changes, it might be a good time to remove the typemaps and introduce a wrapped Variant class. 
Maybe this will happen soon enough, when the first OPC UA pub/sub implementation is released (of which I'm eagerly waiting...)

WimPessemier, 2017-07-20:
Fixed and tested, you can just pull version 2.3.1 from the master. 

The following now works:
```python
import pyuaf
from pyuaf.util import VariantVector
from pyuaf.util.primitives import Float, UInt32

def isEqual(a, b):
    """Compare 2 floats."""
    return abs(a - b) < 0.00001

vec1 = VariantVector()
vec1.append(Float(2.3))
vec1.append(UInt32(33))

vec2 = VariantVector()
vec2.append(vec1)
vec2.append(UInt32(44))

vec3 = VariantVector()
vec3.append(vec2)
vec3.append(UInt32(55))

assert isEqual(vec3[0][0][0].value, 2.3)
assert isEqual(vec3[0][0][1].value, 33)
assert isEqual(vec3[0][1].value, 44)
assert isEqual(vec3[1].value, 55)
```

closed 2017-07-20 by WimPessemier