uaf/uaf #123: PyUAF with LDS-ME

timtarax, 2017-06-20:
Hi Wim,

in my use-case I've installed a local discovery server with multi-cast extension (LDS-ME) from OPCFoundation site. 
Now, I tried to use PyUAF and it's implemented discovery process with the LDS-ME. But I didn't get all servers from the LDS-ME. I got only the servers which are directly registered at the LDS, but no servers which is announced by another LDS in the local network. 

I think the problem is, that PyUAF uses only the foundServers() function of the SDK. But I'am not sure. 
Maybe you can have a look on this?

Thanks!
Tim

WimPessemier, 2017-06-24:
Hi Tim, as mentioned by e-mail I'll look at it in one of the next weeks (after issue https://github.com/uaf/uaf/issues/122 is done).
I have never looked into the multicast extensions, but I guess you need the SDK's [findServersOnNetwork](http://documentation.unified-automation.com/uasdkcpp/1.5.5/html/classUaClientSdk_1_1UaDiscovery.html#a39104c277dd7f00a2200789c80c74532) functionality? Do you think this would solve your problem?

timtarax, 2017-07-11:
Hi Wim, yes the [findServersOnNetwork](http://documentation.unified-automation.com/uasdkcpp/1.5.5/html/classUaClientSdk_1_1UaDiscovery.html#a39104c277dd7f00a2200789c80c74532) solved my problem. 
I've implemented it in your findServers() routine, that the discoveryUrls in the database of the ClientSettings is updated with the results of the LDS-ME on localhost. But I think this solution is not the proper way...

WimPessemier, 2017-07-11:
Hi Tim, OK glad you've got it solved via a quick fix! I'm still busy with https://github.com/uaf/uaf/issues/122 but since I can only spend a few hours per week on it, and since this requires a few structural changes, a definitive solution for your problem will take a few more weeks.
But I'll implement it, no worries :) If possible please share your quick fix (e.g. via https://gist.github.com) so I can make sure the definitive solution suits your needs...

timtarax, 2017-07-14:
Sure, here is the code for modified discoverer.cpp: https://gist.github.com/timtarax/0a307c85cbc71de80799734a438b2ea2
I only add a new function findServersOnNetworksByLocalhost() to update the discoveryUrls in the database and call it in your findServers() before you loop through the discoveryUrls.

WimPessemier, 2017-07-20:
Thanks for sharing your code!
I've finished the previous issue (#122) so I'll implement this new feature in the next days (weeks).

WimPessemier, 2017-08-11:
The feature is added now in version 2.4.0. 

By default, LDS-ME discovery is not enabled. You can enable it and configure it in the client settings:
- [ClientSettings.discoveryOnNetworkEnable](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client_settings.html#pyuaf.client.settings.ClientSettings.discoveryOnNetworkEnable)

You can also run it manually:
- [Client.findServersOnNetworkNow](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#pyuaf.client.Client.findServersOnNetworkNow)
- [Client.serversOnNetworkFound](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client.html#pyuaf.client.Client.serversOnNetworkFound)

If anything is not working as expected, let me know!

closed 2017-08-11 by WimPessemier