uaf/uaf #126: Unable to create correct inputArguments

jorrita, 2017-08-15:
Our server has a single method call that takes nested arrays as follows:
<pre>
Flocks[]
            GrowerName (String)
            CountryCode (Int32)
            GrowerExternalReference (String)
            HouseExternalReference (String)
            FlockExternalReference (String)
            CustomFields[]   (contains 0..5 items of type String)
            Lots[]
                             DisplayName (String)
                             ExternalReference (String)
                             PlannedArrivalTime (DateTime)
                             PlannedAverageWeight (Double)
                             PlannedNumberOfProducts (UInt32)
                             PlannedTotalWeight (Double)
                             CustomFields[]   (contains 0..5 items of type String)
                             LineNumber (UInt32)
                             SequenceNumber (UInt32)
</pre>
(In the model this is defined as a "FlockTypeStruct")

In UaExpert I am able to send an empty  FlockTypeStruct as argument to the method call.
It is however unclear to me how this should be done in pyuaf.
The naive approach of creating a nested VariantVector (as described in issue #122) yields a BadTypeMismatch. Unfortunately, I see no way to get more information from the library then "you made an error":
<pre>
>>> result = self.opcHelper.MeynClient.call(self.UcInterface, methodAddress, inputArgs=[mkFlock()])
>>> result.targets[0]
<pyuaf.client.results.MethodCallResultTarget( - clientConnectionId              : 0
                                              - status                          : The server could not successfully process the MethodCall service: BadTypeMismatch
                                              - opcUaStatusCode                 : BadTypeMismatch
                                              - outputArguments[]               : []
                                              - inputArgumentStatuses[]         : [There was an error for the given input argument: BadTypeMismatch]
                                              - inputArgumentOpcUaStatusCodes[] : [BadTypeMismatch])>

</pre>

In e-mail correspondence, you hinted that ExtensionObjects are the way to go, but I don't see how that should work. In the example https://github.com/uaf/uaf/blob/master/examples/pyuaf/client/how_to_read_and_write_structures.py 
only existing StructureDefinition/ExtensionObjects are used, where I would have to build one from scratch. I have not found a way to do that.

Note: in commit 28c82c46 you have forgotten to add the documentation examples for constructing a StructureDefinition (api_pyuaf_util.rst:3778)

WimPessemier, 2017-08-16:
Hi jorrita, thanks for writing everything down on github.
I'll provide a more complete example soon but can you already run this example (and adapt the NodeId of the method)? 
https://github.com/WimPessemier/uaf/blob/args-support/examples/pyuaf/client/how_to_inspect_method_args.py

jorrita, 2018-06-26:
I might be looking at a problem in de SDK here:
<pre>>>> methodAdr = Address( NodeId("UniversalConnectorService.ImportFlocks", namespaceUri), serverUri )
>>> inArgsAdr = Address( methodAdr , [RelativePathElement(QualifiedName("InputArguments", 0))] )
>>> result = myClient.read( [inArgsAdr] )
>>> target = result.targets[0]
>>> extensionObject = target.data[0]
>>> print extensionObject
 - encodingTypeId : Ns=0='http://opcfoundation.org/ua'|Numeric|298
 - dataTypeId     : Ns=0='http://opcfoundation.org/ua'|Numeric|296
>>> argument = Argument(extensionObject)
>>> definition = myClient.structureDefinition(argument.dataType)
>>> print definition
 - name                : FlockTypeStruct
 - documentation       : 
 - dataTypeId          : Ns=2|Numeric|3019
 - namespace           : http://meyn.com/mlogic/
 - baseTypeId          : Ns=0|Numeric|0
 - childrenCount       : 7
 - children
    - child(0)
       - name          : GrowerName
       - documentation : 
       - typeId        : Ns=0|Numeric|12
       - valueType     : 12 (String)
       - arrayType     : ArrayType_Scalar
    - child(1)
       - name          : CountryCode
       - documentation : 
       - typeId        : Ns=0|Numeric|0
       - valueType     : 6 (Int32)
       - arrayType     : ArrayType_Scalar
    - child(2)
       - name          : GrowerExternalReference
       - documentation : 
       - typeId        : Ns=0|Numeric|12
       - valueType     : 12 (String)
       - arrayType     : ArrayType_Scalar
    - child(3)
       - name          : HouseExternalReference
       - documentation : 
       - typeId        : Ns=0|Numeric|12
       - valueType     : 12 (String)
       - arrayType     : ArrayType_Scalar
    - child(4)
       - name          : FlockExternalReference
       - documentation : 
       - typeId        : Ns=0|Numeric|12
       - valueType     : 12 (String)
       - arrayType     : ArrayType_Scalar
    - child(5)
       - name          : CustomFields
       - documentation : 
       - typeId        : Ns=0|Numeric|12
       - valueType     : 12 (String)
       - arrayType     : ArrayType_Array
    - child(6)
       - name          : Lots
       - documentation : 
       - typeId        : Ns=2|Numeric|3018
       - valueType     : 22 (ExtensionObject)
       - arrayType     : ArrayType_Array
 - isUnion             : False
 - binaryEncodingId    : Ns=2|Numeric|5221
 - xmlEncodingId       : Ns=0|Numeric|0
>>> structureValue = GenericStructureValue(extensionObject, definition)
>>> print structureValue
 - definition
   NULL
 - fields                 : (No fields to display, definition is NULL)
</pre>