uaf/uaf #127: Callback for monitored data in UAF

digitalshow, 2017-09-01:
What is the recommended way to handle data changes in C++ UAF? I can't find an easy way to register a callback method.
I realize that I have to implement the ClientInterface. Should I build a class that inherits the Client class or can I somehow inject an own class into the Client or the factories that the Client creates?

WimPessemier, 2017-09-03:
Hi, the only way foreseen in the C++ code is to inherit from uaf::Client (which inherits from uaf::ClientInterface). You can then override all virtual methods from uaf::ClientInterface.

If you prefer to work with function pointers, you can easily write a generic helper class that does this.
But he best way would be to add some code to uaf::Client, make a pull request, and then we'll support it for everyone :)

Client methods that accept callbacks are currently only implemented in Python (therefore the Python API is more extensive). SWIG is not too good with function pointers, so therefore the C++ API has been kept more basic...

WimPessemier, 2017-09-27:
Very nice, thanks a lot for your contribution!!
I'll try to test everything in the weekend but it looks very good. Did you test this in Windows or Linux?
We should also adapt the changelog and version, feel free to do it, or I can do it as well, whatever you prefer.
Thanks!

WimPessemier, 2017-10-24:
Finished (v 2.5.0), see https://github.com/uaf/uaf/pull/128

closed 2017-10-24 by WimPessemier