uaf/uaf #129: Add ServerTimestamp and SourceTimestamp to DataChangeNotification

ravn244, 2017-10-20:
Currently the DataChangeNotification has no properties for the ServerTimestamp and the SourceTimestamp.
Its important for us to have the values for these for the correct timing because the signal propagation times (Source->Server->Clients) can be different in ms range.

The DataValue type holds the timestamps but I saw thats only used in a part of the HistoryReadRaw results and not also in DataChangeNotification, right?

WimPessemier, 2017-10-21:
Hi, this is fixed in the above commits. I'll test it in one of the next days and then merge it in the master branch.

HimanshuDobaria, 2018-06-12:
Hello Wim,

I have tried above code but getting error in "subscription.cpp:432:108: error: wrong type argument to unary plus" Can you please advise on priority.

------------------- Output ----------------
/root/OPC-UA/uaf-master-st/src/uaf/client/subscriptions/subscription.cpp: In member function ‘virtual void uaf::Subscription::dataChange(const UaMonitoredItemNotifications&, const UaDiagnosticInfos&)’:
/root/OPC-UA/uaf-master-st/src/uaf/client/subscriptions/subscription.cpp:432:108: error: wrong type argument to unary plus
 +               notification.sourceTimestamp.fromSdk(UaDateTime(dataNotifications[i].Value.SourceTimestamp));
                                                                                                            ^
make[2]: *** [uaf/client/CMakeFiles/uafclient.dir/subscriptions/subscription.cpp.o] Error 1

HimanshuDobaria, 2018-06-14:
Hi,

Any update on this, I have tested and it is not compiling.

HimanshuDobaria, 2018-07-05:
Hi, Have you got time to check the issue ?

WimPessemier, 2018-07-05:
Hi, I presume that you copy-pasted the diff in the above commits, but you forgot to remove the '+' in the diff.
See https://www.git-tower.com/learn/git/ebook/en/command-line/advanced-topics/diffs to know more about diff syntax.

HimanshuDobaria, 2018-07-09:
Hi Wim,
Thank you so much, it's working fine now.