uaf/uaf #130: Test test_client_Client_processRequest_some_historyReadRawModifiedRequest_with_manual_continuation  fails

robi-wan, 2017-10-25:
After a recent push one test fails:

```
HistoryReadRawModifiedTest.test_client_Client_processRequest_some_historyReadRawModifiedRequest_with_manual_continuation |  
24 Oct 17 22:08
Failure
Traceback (most recent call last):  
File "C:\development\lang\python27\Lib\unittest\case.py", line 331, in run    
    testMethod()  
File "C:\BuildAgent\work\4e5aa514b5bf26c9\unittests\pyuaftests\client\client_historyreadrawmodified.py", line 153, in test_client_Client_processRequest_some_historyReadRawModifiedRequest_with_manual_continuation    
    self.assertGreater( len(allData) , 2 )  
File "C:\development\lang\python27\Lib\unittest\case.py", line 944, in assertGreater    
    self.fail(self._formatMessage(msg, standardMsg))  
File "C:\development\lang\python27\Lib\unittest\case.py", line 412, in fail    
    raise self.failureException(msg)
AssertionError: 2 not greater than 2
```

One of these commits may have broken the test:

Merge pull request #128 from digitalshow/add_callbacksImplementing callbacks for monitored items | wim pessemier <wimpessemier@users.noreply.github.com> |   | 0 files | 9538e411257f838… | 24 Oct 17 21:28 |  
-- | -- | -- | -- | -- | -- | --
Merge branch 'master' into add_callbacks | wim pessemier <wimpe@ster.kuleuven.be> |   | 5 files | d8dac56cfd8f7cb… | 24 Oct 17 21:04 |  
bump version | wim pessemier <wimpe@ster.kuleuven.be> |   | 3 files | b791be90b2cdacf… | 24 Oct 17 20:50 |  
Added example of registring a static method | wim pessemier <wimpe@ster.kuleuven.be> |   | 1 file | b8180daa781b852… | 21 Oct 17 22:36 |  
modified code to c++98, for backwards compatibility | wim pessemier <wimpe@ster.kuleuven.be> |   | 6 files | 4a68ee21b5be548… | 30 Sep 17 22:11 |  
Implementing callbacks for monitored items | ingo koinzer <ingo@koinzer.net> |   | 7 files | 91b1b897995dc9b… | 06 Sep 17 16:35

Our integration server build and ran pyuaf with Visual Studio 2010 (Windows XP), OPC UA SDK 1.5.3, python 2.7.5, cmake 3.2.x, SWIG 3.0.5

WimPessemier, 2017-10-27:
Hmm I'll re-run the tests but I see no logical connection between the commits and the failure. Is this repeatable?

robi-wan, 2017-11-01:
I re-run the build (including the tests) again (on the original build agent and on another build agent of our CI system): all tests passed.

closed 2017-11-01 by robi-wan

WimPessemier, 2017-11-01:
OK thanks! If it happens again make sure to report it!