uaf/uaf #132: manuallyConnectToEndpoint() doesn't use sessionSettings

jorrita, 2018-01-05:
It looks like there is a problem in uaf/src/uaf/client/sessions/session.cpp - any session settings that are given in the call are ignored. Looking at the code, the line 
<pre>
    ret = initializePkiStore(uaSecurity);
</pre>
seems to unreachable.

Here is a possible patch:
<pre>
--- a/src/uaf/client/sessions/session.cpp
+++ b/src/uaf/client/sessions/session.cpp
@@ -566,8 +566,7 @@ namespace uaf
         // ============
 
         // initialize the PKI store so that we can verify the server certificate
-        if (ret.isGood())
+        if (connectionAttemptStatus.isGood())
            ret = initializePkiStore(uaSecurity);

         // load the server certificate
</pre>

WimPessemier, 2018-01-06:
Hi,
Indeed this is a bug. I think your patch will work, but it's probably cleaner to 
 - remove the if statement, and always execute ``ret = initializePkiStore(uaSecurity);``
 - remove the declaration of ``connectionAttemptStatus`` (and replace its usages by ``ret``) in the ``connectToSpecificEndpoint`` function since it is not needed.

I'll try to find some time for a pull request (or just make one if you want)

jorrita, 2018-01-08:
Agreed.