uaf/uaf #134: Getting null value of GenericStructureValue object when browse the Extension Object like "EngineeringUnits" & "Range"

HimanshuDobaria, 2018-06-06:
Hello Team,
I want to browse "Demo.BoilerDemo.Boiler1.FillLevelSensor.FillLevel.EngineeringUnits" and Demo.BoilerDemo.Boiler1.FillLevelSensor.FillLevel.Range" node, both are type of extension object, I am using the example given in "how_to_read_and_write_structures.py" but I am getting null value in below code.

structureValue = GenericStructureValue(extensionObject, definition)

I am getting null value of "structureValue", I have printed extensionObject and definition both are proper as per expected.

Anybody having an idea how to get "EngineeringUnits" and "Range" data type ?

HimanshuDobaria, 2018-06-08:
Any update ? How to read "EngineeringUnits" and "Range" property of node

WimPessemier, 2018-06-08:
Not yet, I'll take a look at it over the weekend.

WimPessemier, 2018-06-09:
Hi,
support for engineering units and ranges is now added. Apparently, GenericStructureValues are not so generic after all, so I had to wrap the pyuaf.util.Range and pyuaf.util.EUInformation classes.

See
- the example https://github.com/uaf/uaf/blob/master/examples/pyuaf/client/how_to_read_engineering_units_and_ranges.py
- http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_util.html#class-euinformation
- http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_util.html#class-range

This should fix the issue.

closed 2018-06-09 by WimPessemier

HimanshuDobaria, 2018-06-11:
It's working fine. Thank you so much for quick reply and implementation. Great support. It helps us a lot.

HimanshuDobaria, 2018-06-11:
Hi Wim,

Thank you so much for quick implementation, really appreciate for this
great support. :)

Thanks,
Himanshu