uaf/uaf #137: Need "setWhereClause" support in EventFilter class

HimanshuDobaria, 2018-06-19:
Currently in "EventFilter" class there is no support of where clause it is very important to provide this support. 

Please provide this support

WimPessemier, 2018-06-19:
Hello,
it's quite some work, but I can guide you if needed:

1. Wrap UaLiteralOperand, and UaElementOperand.
2. Wrap the OpcUa_FilterOperator_... enumerator
3. Wrap UaContentFilterElement and UaContentFilter
4. Add a whereClause member to EventFilter
5. Include the whereClause in the service invocation (invocations/createmonitoredeventsinvocation.h)

I don't have the time in the next months to implement features like these, so if you need it, you'll have to do it yourself. I can assist with more details, of course.

HimanshuDobaria, 2018-07-05:
Hi,
Can you assist in more details ?

WimPessemier, 2018-07-05:
Hi,

to start with step 1:

- Look at simpleattributeoperand.h and simpleattributeoperand.cpp in https://github.com/uaf/uaf/tree/master/src/uaf/util, and wrap UaLiteralOperand and UaElementOperand in a similar way. 
- Add the lines in https://github.com/uaf/uaf/blob/master/src/pyuaf/util/util_util.i just as for simpleattributeoperand.
- Add unit tests in https://github.com/uaf/uaf/tree/master/unittests/pyuaftests/util
- Document your new classes in https://github.com/uaf/uaf/blob/master/src/pyuaf/doc/api_pyuaf_util.rst.
- Push your branch, and then I'll explain step 2.