uaf/uaf #138: Permissions of Server

YarochkinMichael, 2018-08-22:
I need to design system with user assess management : different Users have access to different Nodes from different Clients.
Best is Role based. But my task is to use what I can find on a market.

So, my question is how far I can go with your implementation?

Here is one example of what I want to have.
It is not real, but close to reality.

Users:
U1 :operator
U2 : supervisor
U3 : maintenance

Groups:
G1 :operators
G2 : supervisors
G3 : maintenance

Clients:
C1 : terminal at machine
C2 : terminal at office

Servers:
S1 : machine

Nodes:
N1 : status
N2 : product counter
N3 : product reset counter, new shift
N4 : bad product counter
N5: reset bad product counter

Access:
U1->  G1->  C1->S1->N1, N2, N3, N4
U1->  G1->  C2->S1->N1, N2, N4

U2-> G2-> C1->S1->N1, N2, N4
U2-> G2-> C2->S1-> no access

U3-> G3-> C1->S1->N1, N4, N5
U3-> G3-> C2->S1->N1, N4

WimPessemier, 2018-08-25:
I don't fully understand your question, but in UAF (client only!) [you can login with user credentials](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_client_settings.html#pyuaf.client.settings.SessionSecuritySettings) and [you can easily read which variables are readable/writeable and which methods are executable by the user](http://uaf.github.io/uaf/doc/pyuaf/api_pyuaf_util_attributeids.html#module-pyuaf.util.attributeids).

YarochkinMichael, 2018-08-27:
This is task for Server, indeed.

So, you have only Client now?
Any plans for Server?

WimPessemier, 2018-08-27:
Indeed, only client support. No plans for server support.