uaf/uaf #139: Permissions on Client

YarochkinMichael, 2018-08-22:
I want to design "Universal OPC UA Client".
It is an ideal solution.
Goal is to make something "now".
And evolve later.

-on a startup is only prompt with User credentials
-credentials are send to GDS for validation
-GDS contacting company LDAP and retrieving access level
-GDS is sending back to Client:
      -Server Address Space, accessible to User
      -UI Schema, for this User on this Client
     -Access Token based on User group and Clinent
-Client:
     -configuring UI
     -Using Access Token to connect with Server
-All buttons and indicators are working according UI Schema
-Everybody is happy :)

Do you like this journey?

Your dreamer, Mikl

WimPessemier, 2018-08-25:
Hello Mikl,

Feel free to build further on UAF to create your "ideal" solution 👍 
Bear in mind that UAF is currently supporting client-side only.
Feel free to ask for advice or for pointing you in the right direction.

Wim

YarochkinMichael, 2018-08-27:
:)

It will be interesting for you to develop it under our specs?

Mikl