uaf/uaf #141: Python 3 compatibility

WimDeMeester, 2019-06-21:
Hi @WimPessemier,

I think that uaf is not compatible with python 3. Is that correct? 
Can you estimate how much time it will cost to convert to python 3 (let's say for a jobstudent)?
We could try to find someone to work on the python 3 conversion.

khalimec, 2020-02-10:
Hi,
It looks like that pyuaf is pretty much compatible with python 3 (when compiled with python 3 libs).

Traceback of the errors we have got when running pyuaf with python3:
`Traceback (most recent call last):
  File ".../python3.7/site-packages/pyuaf/client/requests.py", line 1487, in __getElementFromKwargs__
    if kwargs.has_key(key):
AttributeError: 'dict' object has no attribute 'has_key'
...
During handling of the above exception, another exception occurred:
...
Traceback (most recent call last):
  File ".../python3.7/site-packages/pyuaf/client/__init__.py", line 1177, in __del__
    ClientBase.__del__(self)
AttributeError: type object 'ClientBase' has no attribute '__del__
...
During handling of the above exception, another exception occurred:
....
Traceback (most recent call last):
  File "../my_application/cores/opcuaconnection.py", line 145, in opcUaClient
    except (SecurityError, PathNotExistsError) as error:
TypeError: catching classes that do not inherit from BaseException is not allowed`

Below are the issues that should be fixed:
1. In _src/pyuaf/client/client_requests.i_
_line: 123_ 
`if kwargs.has_key(key):` 
should become 
`if key in kwargs:`
2. In _src/pyuaf/util/util_errors.i_
_line: 57_  
`%pythoncode %{ _object = Exception %}` 
should become 
`%pythoncode %{ object = Exception %}`
Or, the generated (created) `class UafError(object)` 
should be 
`class UafError(_object)`  the underscore before "object" was missing :-)!
3. In _src/pyuaf/client/client.py_
_line: 86_  
`ClientBase.__del__(self)` should be removed, it is not needed.
This is because, destructors are called when an object gets destroyed. In Python, destructors are not needed as much needed in C++ because Python has a garbage collector that handles memory management automatically.
In python the __del__() method is called when all references to the object have been deleted i.e when an object is garbage collected.

_# Python programs to illustrate destructor_
`class Employee: 
    # Initializing 
    def __init__(self): 
        print("Employee created") 
    # Calling destructor 
    def __del__(self): 
        print("Destructor called")
 
def Create_obj(): 
    print("Making Object...") 
    obj = Employee() 
    print("function end...") 
    return obj 

print("Calling Create_obj() function...") 
obj = Create_obj() 
print("Program End...")`
_Output:_
`Calling Create_obj() function...
Making Object...
Employee created
function end...
Program End...
Destructor called

`class A: 
    def __init__(self, bb): 
        self.b = bb 

class B: 
    def __init__(self): 
        self.a = A(self) 
    def __del__(self): 
        print("die")

def fun(): 
	b = B() 

fun() `
_output:_
`die`

jorrita, 2020-05-20:
I see also a lot of 'xrange' in the python code, which is 'range'  in python3.

tvroegop, 2020-05-26:
> 2. In _src/pyuaf/util/util_errors.i_
>        _line: 57_
>        `%pythoncode %{ _object = Exception %}`
>        should become
>        `%pythoncode %{ object = Exception %}`
>        Or, the generated [...]

Actually Exceptions should derive from BaseException at a minimum in python3.

jorrita, 2020-05-26:
JA, die staat al in dat issue

tvroegop, 2020-05-26:
in file src/pyuaf/util/util_primitives.i the references to 'long' should become 'int' since long does not exist anymore in python3.

in file src/pyuaf/client/client.py the following functions need to change (not python3 related by the way)
1. __dispatch_untrustedServerCertificateReceived__ needs a correct return statement in the exception clause. I suggest pyuaf.util.PkiCertificate.Action_Reject.
2. __dispatch_connectErrorReceived__ needs a correct return statement in the exception clause. I suggest False which coincides with default action.